prusa3d/PrusaSlicer #7978: gcode substitution: replacing comments removes initial and final M73 commands

fe60, 2022-02-24:
### Version
PrusaSlicer-2.4.1-beta1+win64-202202010918

### Operating system type + version
Windows 10 (21H2)

### Behavior
I try to remove comments from a gcode file using gcode substitution (#7029).
Search patterns: `^;.*\n` and `;.*`  (Regular expression: On, Match single line: off)
Replace with: nothing
The patterns work as expected, except from the removal of the initial and final `M73` commands.
From the original sample file `simple.gcode` the following commands are removed unexpectedly:

line 12 and 13
```
M73 P0 R0
M73 Q0 S0
```
line 587 and 588
```
M73 P100 R0
M73 Q100 S0
```

* _Expected Results_
Only comments are removed from the gcode.
* _Actual Results_
Initial and final `M73` commands are removed.
The problem occurs even when using only one of the search patterns mentioned above (no matter which one).

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[PrusaSlicerReplaceComments.zip](https://github.com/prusa3d/PrusaSlicer/files/8131024/PrusaSlicerReplaceComments.zip)

Archive contains a folder with three files:
* project file with simple example
* original gcode of simple example
* gcode generated with substitutions

bubnikv, 2022-02-24:
see https://github.com/prusa3d/PrusaSlicer/issues/7955#issuecomment-1046708780

fe60, 2022-02-24:
https://github.com/prusa3d/PrusaSlicer/issues/7955#issuecomment-1046708780
> The G-code is generated first, then post-processed with the regexp find / replace post-processor, then processed with a firmware simulator which estimates times. [...] Thus the find / replace post-processor cannot replace M73 lines.

According to this statement, the M73 commands have not been replaced, but the substitution/removal of comments somehow prevents the insertion of the initial and final M73 commands.

bubnikv, 2022-02-24:
Removing all comments is not a good idea in general, as some of the comments are used internally by PrusaSlicer to augment G-code understanding by the G-code viewer. Without these comments the G-code the layer height or extrusion type are not known.

These are the reserved tags stored by PrusaSlicer as comments, which you are inadvertently stripping off:
    "TYPE:",
    "WIPE_START",
    "WIPE_END",
    "HEIGHT:",
    "WIDTH:",
    "LAYER_CHANGE",
    "COLOR_CHANGE",
    "PAUSE_PRINT",
    "CUSTOM_GCODE",
    "_GP_FIRST_LINE_M73_PLACEHOLDER",
    "_GP_LAST_LINE_M73_PLACEHOLDER",
    "_GP_ESTIMATED_PRINTING_TIME_PLACEHOLDER"

Namely, the last three comments drive the M73 generator, which runs after the find/replace post-processor.

foreachthing, 2022-02-24:
> _GP_LAST_LINE_M73_PLACEHOLDER

Why not rename M73 to M-7-3 (or something else),so regex won't find it?

fe60, 2022-02-24:
Now I got it. It's at least a three step approach:
* gcode is generated with some placeholders marked as comment
* find / replace post-processor is executed
* some placeholder inserted in step one are replace with commands (e.g.: `;_GP_[FIRST|LAST]_LINE_M73_PLACEHOLDER`)

My substitutions (Search patterns: `^;.*\n` and `;.*`) removed all comments including the placeholders for the initial and the final M73 command. As a result these commands have not been inserted by the M73 generator.

While removing comments breaks the G-code viewer, other users want to remove the comments to get the gcode working on certain 3D printers (https://github.com/prusa3d/PrusaSlicer/issues/7029#issuecomment-931185374). I just tried to get this done using the brand new substitution feature.
For me it is now clear, what's going on and the issue can be closed.

Mentioning this behavior in the [documentation](https://help.prusa3d.com/en/article/g-code-substitutions_301694) could help to avoid such confusion.
It should also mention the substitution
> Neither [touches] the initial comments at the start of the G-code nor the trailing config section

https://github.com/prusa3d/PrusaSlicer/issues/7952#issuecomment-1046692603

bubnikv, 2022-02-25:
I have extended the blocks suppressed by the G-code find / replace algorithm to cover these M73 placeholders and all the trailing comments generated by PrusaSlicer.

fe60, 2022-03-03:
Works as expected with 2.4.1-rc1.

closed 2022-03-03 by fe60