prusa3d/PrusaSlicer #7981: 2.4.1-beta3 crashes on launch

DrLex0, 2022-02-24:
### Version
2.4.1-beta3

### Operating system type + version
Mac OS 10.14.6 (18G9323) (Mojave)

### 3D printer brand / version + firmware version (if known)
Doesn't matter

### Behavior
* Doesn't even get to the splash screen.
* [Crash_report.txt](https://github.com/prusa3d/PrusaSlicer/files/8134175/Crash_report.txt)

bitnautist, 2022-02-25:
Crashes for me as well under macOS 10.14.6 with Prusa i3 MK3s+ Firmware: 3.10.1-4697

lukasmatena, 2022-02-25:
@DrLex0 @bitnautist 
Thanks for the report. There seems to be a problem with older macOS versions, I also have issues when I tried on 10.10, although the crash report look different. We managed to get it running, but we fear that your issue might be independent. Could you both please try out the following build and report if anything changed? Thanks.

[PrusaSlicer-2.4.1-beta3+1-universal-g97fb6d22d](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/lm/PrusaSlicer-2.4.1-beta3+1-universal-g97fb6d22d-202202251221.dmg)
user: `prusaslicer`
password: `slicer`

DrLex0, 2022-02-25:
@lukasmatena This build works for me 🎉

PhilippeGalid, 2022-02-25:
Thanks the PrusaSlicer-2.4.1-beta3+1-universal-g97fb6d22d on MacOs 10.14.6 works now, Thanks for your fast Support.
Thanks for your hard work to make the best Slicer

K2Kevin, 2022-02-25:
Works!

closed 2022-03-24 by lukasmatena