prusa3d/PrusaSlicer #7983: Parametric settings

OfficialDyray, 2022-02-24:
The problem: 

I often make small tweaks to the current profile I am working on and the changes aren't propagated to the many other profiles I have. Another problem is that I have many different profiles for every use case. Such as printing fast/slow and durable/filament-saving.

Example: I have a general printing profile, a fast profile, and a strong profile. In the general profile, I change my support overhang angle. If I want that setting to change in the other profiles I have to go through manually and switch them one at a time.
 
Another Example: I have a profile for printing quickly, and one for printing strong. I have to make another profile to combine the two to print strong and quickly. 

A solution: 

I would benefit if profiles could be applied parametricly. You would have a base profile that contains the default settings for your printer. Then a fast profile could be applied on top of that profile to change the fan speed, acceleration, and print speed. The settings could overwrite, or do math operations on the old values(i.e. double the base fan speed). Then another profile could be applied after that one for strength. Such as increasing the wall count/infill density. 

Other benefits:

Greatly reduces clutter and simplifies maintaining multiple different profiles.

Area5142, 2022-02-25:
This would be great.
I often change the same settings depending on the model I want to print. Could be to change the number of perimeters, top and bottom layer or setting elephant food.
All on a standard base profile, often the one PR provides.

I see it like running a small macro after selecting the default profiles, where only the differences is defined. I could be something like:

```
   ; Example: Setup for fast print
   ; Updated: 2022.02.25
   Perimeters=2 ; Set perimeter count
   Solid_Layers_Top=4 ; Set top and bottom layers
   Solid_Layers_Bottom=3
   Fill_Gaps=off
   Elephant_Food_Compensation=0 ; Off
   Speed_External_Perimeters+=10 ; Raise speed with 10mm/s
   Bridges_Fan_Speed=60% ; Raise fan speed by 10%
```
Operators like =, +=, -=, *=, /= could be used to make simple math (like in programming languages like C). Everything after the ";" is treated as a comment, like in GCode.
The setup could be saved in a file and loaded/run from the dialog. It will change the current defined setting. Saving everything in a separate file will give some advantages, like: Filename is the macro name and it is easy to edit, clone and share.
This is an expert setting as the result could give illegal settings.

To help write a command line in the macro, right clicking a setting could offer to generate the text for the current setting and place it on the clipboard for inserting into a macro file (editor).

Just some thoughts...

OfficialDyray, 2022-02-25:
I like the idea of having the modifier/macro laid out like a programming language. For the gui there would have to be some type of ordering menu where you chose what macros/modifiers to use and what order they will be in. Maybe have a plus to add macros/modifiers, then have an X and movement handle on each to order them from first to last. 

We would have to decide what to do with data types, do we use 0% - 100% for fans or 0.0 - 1.0? Would we allow setting the temperature of the hotend parametrically (might be useful)? We also need to consider a way to set the maximum and minimum range for every value so that it cant go outside of a certain range(cant force the speed too low or too fast).

User definable variables could be provided in the form of sliders/text fields:
```
User_Slider = 0-1 ; slider with range 0 to 
Fan_Speed = User_Slider ; set fan speed to slider
Something_Else = User_Slider ; set something else
```

You could go potentially very advanced by providing different variables to use.
Example: Possibly be able to set the fan speed based on the current layer time. (the fan speed proportional to the layer time)
`Fan_Speed = 100 - Layer_Time * 10 ; fan proportional to layer time` ( less than 10 seconds ramps up to 100% speed)

The last two mentions are a bit ambitious, but would be very cool to implement.

foreachthing, 2022-02-25:
Like @Area5142 said! Just need a textbox to enter those "globally overriding commands".
No fancy GUI, just plain text (with autocomplete ;-) ).

bubnikv, 2022-02-25:
Similar functionality is implemented in Cura in their configuration back-end. That is not such a big deal to do. The problem is how to present the formulas to the user. Cura just presents the calculated numbers to the user in the UI. If the number is modified by the user, it is saved as such into a user profile.

We are actively looking into such a system to support multi-extruder priting with differing nozzle diameters. However we are not sure yet whether this is the best way to go.

Area5142, 2022-02-25:
I agree with @foreachthing, this should be a simple solution without fancy GUI. Just a set of overrides in a file (macro) that can be run on the current loaded setup.
With simple rules, like format and range checking done the same as when entering the values manually. New addition are the operators +=, -= *= and /= to change values based on the previous value.
It is an expert setting like the GCode substitution with regular expressions.

Depending on user response and popularity a more polished version could follow in a later release, supporting a more user friendly interface.

A simple tool like this would be very useful in my toolbox. I see many cases where I could use a macro on standard profiles:

- Setup for fast prototype: 3-2 top bottom layers, 1 perimeter low fast infill and speed increase.
- Setup for SL1S model directly on build plate: Lower first layer exposure time, set slow/fast mode etc.
- Setup for detailed model: Activate detect thin wall, avoid crossing perimeter, Seam position Aligned and Elephant_Food_Compensation=0.2.
- Setup for larger model: Perimeters=3, 6-5 top bottom layers, Z_Lift=0 (PETG).

Today it is only possible to make a cross (Print, Filament, Printer) setup using config bundles and I think they are hard to maintain as it is not obvious what is changed and why, compared to the standard profile. I know that I can compare profiles in the GUI, but I have to load the bundle and there is no comments to the changes as there would be in a macro.

OfficialDyray, 2022-02-25:
One feature that I'd like to emphasize is being able to layer different macros
[image]

Area5142, 2022-02-26:
@OfficialDyray this seems to me to be a little complicated to set up and manage.

To avoid managing inheritance (hierarchy) I would suggest just running more macros in the right order after each other to achieve the same effect.

1. Run: Base material macro (PLA, PETG, ...)
2. Run: Speed macro
3. Run: Strength macro

When running more macros that changes the same parameter the last macro run will be the winner.

Not defining any structure for the macro system is its strength - you can use it how you like. To make simple changes or complex ones involving print, filament and printer settings - it is up to you.

OfficialDyray, 2022-02-26:
You have a point. I guess I have a tendency to over complicate things. Swapping out macros would be fast enough that you wouldn't need to have every different option predefined. You would just have to change the macro to a different one rather than making a whole new tree.

Area5142, 2022-02-26:
I would suggest a bare minimum GUI as a menu entry, pointing to a list with the names of the files. The name of the macro reflect what it does.
The list has "run", "new", "delete", "edit" and "cancel", where "new" and "edit" just invokes an external editor like Notepad, Wordpad or similar.
All files are located in a single directory under PrusaSlicer configuration and follows the rules for configuration files (Configuration snapshots).

closed 2026-09-29 by prusaslicer-bot[bot]