prusa3d/PrusaSlicer #7986: [Feature Request] Single Perimeter top layers

Weskus0203, 2022-02-25:
### Version
2.4

### Operating system type + version
Windows 10 21H1
OS 19043.1526

### 3D printer brand / version + firmware version (if known)
Voron 2.4 running klipper. No impact on request.

### Behavior
Recently switched from SuperSlicer to PrusaSlicer and one feature I really miss having is single perimeter top layers. I makes a really clean looking part. No matter the number of perimeters selected the option will always print the top layer with a single perimeter. 

[image]

_Is this a new feature request?_
Yes
#### Project File (.3MF) where problem occurs
N/A

cmgreyhounds, 2022-11-27:
Just for reference you can do that currently using a modifier, though I would prefer to have it as a setting. Make a new modifier and select "Layers and Perimeters" for it, then it lets you alter the number of perimeters.

Even BambuLab has this a Single Perimeter Top Layer option in their slicer, and it has very few options.

stepikovo, 2022-12-15:
I'm closing this as a mentioned duplicate of #2151, thx @dartrax

liftbag, 2023-01-28:
I feel like commenting further. Issue 2151 requires, without explicitly specifying it, the possibility of having a different number of layers for the bottom and top infill areas. It is an open issue, but stopped in April 2019. Personally I consider the request mandatory. I rarely use PrusaSlicer due to the lack of this parameter. Sometimes, when a new PrusaSlicer feature hasn't been merged into SuperSlicer yet, I use the height range modifier, but it's frustrating having to use a cumbersome procedure when other slicers just put simple checks.
I hope that the closure of this issue, very specific on the subject, is consequent to the implementation of the single perimeter on top and bottom layer in the next 2.6 release.

Imburr, 2023-04-30:
This is a bit painful to do with modifiers, specially on plates with multiple models and multiple top layers. Would love to see this integrated into PS!

Matszwe02, 2023-05-20:
It will be great if it will be implemented in 2.6.0, if possible. It will make top surfaces so much prettier without the need of ironing. Maybe some developer will read this and implement from superslicer / ask the right team to consider doing it / tell if it's an easy feature that could be likely added in the near future :)

Ibot-11, 2023-06-21:
When can we expect an implementation of this? Seems like A LOT of people want this…
Sadly this gets just ignored for years now…
Same thing with the saved seam preview.
It would be nice to get features added like this instead of ones nobody really asked for.

Zerufuru, 2023-06-25:
Came here to request this exact feature to be implemented a la SuperSlicer.  Single perimeters on the top layer is one of two big dealbreakers that is keeping me from switching to PrusaSlicer.

mojeto, 2023-07-04:
Relevant from https://github.com/prusa3d/PrusaSlicer/issues/10913#issuecomment-1609484522

> Closing as a duplicate. We are planning to add this feature. 

yes please, I miss this feature in PrusaSlicer.

YanceyA, 2023-11-22:
I am hoping this can be a feature to be implemented soon.

tanatos91, 2023-11-23:
Very necessary feature for me too!

SohrabRo, 2023-12-17:
I have been using superslicer only because of this feature. Despite the updates prusaslicer is getting this most have feature that is already available in most of the slicers is missing. I really beg the developer team to add this and let us free of the hassle of switching slicers only for the "only one perimeter on top"

BjoernSch, 2023-12-20:
Please add this feature.
Modifiers are painful with different objects and multi-level top surfaces.

chrishilbert, 2024-02-15:
+1. I just opened an issue on this that I need to close.

nakwada, 2024-02-20:
Would love to have this option in PS, for top AND mostly for bottom surfaces as I do a lot of multicolor prints.
It's available in OrcaSlicer as well.

benvrakas, 2024-04-15:
Want this so badly you can't even imagine. Nice to see the prusa $ at work to add good features.

lukasmatena, 2024-06-25:
Implemented since 2.8.0-alpha5. Closing.

closed 2024-06-25 by lukasmatena