prusa3d/PrusaSlicer #7987: macOS 2.4.0+arm64 generates “Loading of a model file failed.” error

DominikHoffmann, 2022-02-25:
### Version
2.4.0+arm64 for macOS

### Operating system type + version
macOS Monterey 12.2.1

### 3D printer brand / version + firmware version (if known)
Prusa MK3S+

### Behavior
Using File→Import…→Import STL/OBJ/AMF/3MF… results in an error alert saying, “Loading of a model file failed.”

The file was created in Inventor 2022. Both the IPT and STL files have been linked. A screenshot of the alert is inserted below.

https://photonautomation-my.sharepoint.com/:u:/p/dhoffmann/EQHxXfgPyUtBne1Xwhmsq18BKiqxKLQUgrapE9GSG7JqvQ?e=pnIPIw
https://photonautomation-my.sharepoint.com/:u:/p/dhoffmann/EVD0r0Anc1JHrsCfcF_mv10B920rpVt5t1bVA2Num7PnPQ?e=HeJdsp

Not a feature request.

#### Project File (.3MF) where problem occurs
N/A

[image]

FidelCapo, 2022-03-01:
Unfortunately, I am unable to reproduce this issue. Loading the attached STL on my Macbook Air M1 works without problems..

FidelCapo, 2022-03-01:
I have version 12.0.1 (21A559) installed. I will try to upgrade to the latest version.

RichardAlpert, 2022-03-31:
Hello, I have the same issue with Prusa Slicer 2.4.1 on my M1 Mac and I think it is related to the latest Mac OS Update (now running 12.3). 
I made a downgrade to Prusa Slicer 2.3 but the problem is the same. The error message comes up regardless using Drag and drop or via Menu path.

I have no clue and hope the Problem can be reproduced.

RichardAlpert, 2022-04-01:
Today Mac OS 12.3.1 has been released and did the update but it doesn't solve the Problem with Prusa Slicer. Still not able to open files. I already repaired the permissions on the Mac and checked the permissions in the Data Security Center but no success, Prusa Slicer wont open the Files and is stating with the error message shown above as screenshot.

RichardAlpert, 2022-04-01:
So, I wiped off PrusaSlicer completely including all the content in the hidden application Folder, downloaderd 2.4.1 again, installed and still it is not possible. I have no clue what else I could check....

lukasmatena, 2022-04-07:
@DominikHoffmann I'm closing this issue and keeping the duplicate #8151 open. We already started to dig into it there, it would be good to have it all in one place. If you want to try the same I asked @RichardAlpert to do in https://github.com/prusa3d/PrusaSlicer/issues/8151#issuecomment-1091608020, you are most welcome. In that case, report your findings in #8151. Thank you.

lukasmatena, 2022-04-07:
@DominikHoffmann The duplicate issue has taken an unexpected turn: https://github.com/prusa3d/PrusaSlicer/issues/8151#issuecomment-1091848443. Could you please check that there is not the same problem in your case and that the file you have locally on the computer and are trying to open contains something looking like real data? Thanks.

bubnikv, 2022-05-17:
There was no response for more than a month. We hope this issue is a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/8151#issuecomment-1091848443
closing

closed 2022-05-17 by bubnikv