prusa3d/PrusaSlicer #7988: Easy customization of first layer's perimeter settings

joshcarter, 2022-02-25:
### Version
2.4.0 (but applies to all)

### Operating system type + version
MacOS 11.5.1 (but applies to all)

### 3D printer brand / version + firmware version (if known)
Prusa Mk3s (but applies to all)

### Behavior
I've used a trick ever since PrusaSlicer supported height modifiers to get clean first layers, but it's time consuming and I believe that what I'm doing manually would be beneficial as a built-in slicer option. As others have noted (examples: #5650, #2846) short extrusions on the first layer are problematic, and there's great benefit to doing outside perimeters first. I can't think of any reason, really, to do inner perimeters first while on the first layer.

This is what I do manually:

* Create a height range modifier that applies to the first layer only, for example start: 0, end: 0.2mm, layer height: 0.2mm.
* Add option "external perimeters first" and enable it.
* Add option "perimeters" and set to one. (Gives first layer a cleaner appearance.)
* Create a height range modifier that applies to the second layer to 1mm, for example start: 0.2mm, end: 1.0mm, layer height: 0.2mm
* On this second height range modifier, add option "ensure vertical shell thickness" and enable it. If I don't add this, I've noticed that PrusaSlicer will start its infill on the second layer regardless of how many bottom solid layers I have configured in the print settings. (This appears to be a bug. It's been present for several versions.)

Doing this for every single model is tedious but eliminates problems with tiny, unanchored extrusions on the first layer. The result prints well, stays adhered to the bed, and looks clean. It's simple and effective. To generalize my request, it would be great if PrusaSlicer had an option to customize perimeter settings for the first layer only, in the same way that you can currently customize the first layer's height and extrusion width.

[image]

_Is this a new feature request?_
Yes.

fe60, 2022-03-01:
Potential workaround: Add and configure `Height range Modifier`(s) once and Copy&Paste to other objects.
Neither `Copy` nor `Paste` is offered in the context menu of the `Layers` and `Range` nodes in the object list, but it actually works (similar to modifiers https://github.com/prusa3d/PrusaSlicer/issues/7152#issuecomment-988810446). 
You can use keyboard shortcuts (`Ctrl+C|V`) or menu entries (`Edit` => `Copy|Paste`).

foreachthing, 2022-03-01:
Or, save the modifier stack to reuse it later.
Like "Add Shape -> Gallery", one could add different (previously saved) modifier stacks to objects.

neophyl, 2022-03-01:
Its what some of us have been doing for commonly used settings.  You essentially end up with a series of small objects , marked as non printing, placed off to one side of the build plate.  Each object becomes a repository of your common settings.  You then save the project as for example 'Default'.  
The you can open up a copy of default and have access to your common settings.  By saving the project you also have your favourite print/filament/printer combo already selected too.  
Useful when you have multiple printers as you can load Default CR10, Default MK3 etc.

joshcarter, 2022-07-28:
@fe60 thanks so much for pointing out that height range modifiers can be copy/pasted! That saves a huge amount of time. I still think it would be helpful to have some first-layer-only customizations, especially external perimeters first, but in the meantime I'll copy/paste the height range modifiers that I've been using.

closed 2026-09-29 by prusaslicer-bot[bot]