prusa3d/PrusaSlicer #7994: Avoid crossing perimeters crosses external perimeter sometimes on cylinders

gary-wlg, 2022-02-27:
### Version
2.4.1-beta3

Build: PrusaSlicer-2.4.1-beta3+x64-202202241023

### Operating system type + version
macOS 11.6

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S+ stock printer running on latest firmware

### Behavior
While 2.4.1-beta3 greatly reduces crossing of perimeters, there are still cases where the nozzle is travels along the outside of the perimeter, rather than in the centre or on the inside. This can produce a bad finish in this area, especially when printing flexible filament.

In this instance the travel move is the very first move on the layer, to position the nozzle at the start position for the layer.

[image]

gary-wlg, 2022-02-27:
I've attached a slightly different model showing the same issue.
[image]

[CrossPerimeterIssue.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/8148863/CrossPerimeterIssue.3mf.zip)

hejllukas, 2022-03-07:
Thank you for reporting this issue. It should be fixed with 6c51e5148c0fca15c869d344e081aad249b0244f.

closed 2022-03-07 by hejllukas