prusa3d/PrusaSlicer #7996: Hi, up to now I have been working with Slic3r v1.3.0 with no problem. I install the PrusaSlicer 2.4.0 for Windows (actually W10). My object are designed with Solidworks and generate .stl files. With the new vbersion the recognized scaling sems to be incorrect. I join a little objet whidh has a 22mm diameter which is decoded as aa 660mm !

Papyboni, 2022-02-27:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2022-02-28:
Please provide your stl (zip it first).

foreachthing, 2022-03-01:
One comes to mind, while exporting STL from SolidWorks: have you set your units correctly?
Go to [Options...] in the File -> Save As... dialog and make sure your units match. Then also confirm [mm] or [inches] in PS.
Otherwise I've never had any problems and I do nearly all my modeling in SolidWorks 2020...
Maybe you can attach your SolidWorks file(s) as well - zip it/them first ;-)

rtyr, 2023-09-12:
No response, no similar reports. I am closing the issue.