prusa3d/PrusaSlicer #7998: Is Bridge Infill part of support?

laptop-laptop, 2022-02-27:
### Version
2.4.1-beta3

_Use `About->About PrusaSlicer` for release versions_
2.4.1-beta3+win64

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
i3MK3S+

### Behavior
Hi, could developer please clarify if the Bridge Infill layer which is immediately above the Support Material Interface part of support or part of the 3D model? If it is the latter case, I guess it is called Overhang layer? Why the pattern of this layer is made of lots of parallel horizontal lines rather than like the layer above it?

_Is this a new feature request?_

If it is part of the 3D model, please make it looks like the layer above it.

rtyr, 2022-02-28:
The bottom most part of the supported object is printed as a bridge, therefore bridging logic is applied (flow, cooling, speed, angle).

closed 2022-02-28 by rtyr

laptop-laptop, 2022-02-28:
Thank you. Just to clarify, do you mean Bridge Infill layer is part of the 3D model?

rtyr, 2022-02-28:
Yes