prusa3d/PrusaSlicer #8001: configuration wizard

bianrui, 2022-02-28:
### 版本 
version_2.4.1-beta3 包括2.4.1 之前的版本。

### 系统版本+硬件信息

版本	Windows 10 企业版
版本号	21H2
安装日期	‎2021/‎6/‎21
操作系统内部版本	19044.1526
体验	Windows Feature Experience Pack 120.2212.4170.0

设备名称	CreatBot-APP
处理器	Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz   3.70 GHz
机带 RAM	16.0 GB (15.9 GB 可用)
设备 ID	39456730-8DA7-4A13-9689-BDD0D0B270F8
产品 ID	00329-00000-00003-AA432
系统类型	64 位操作系统, 基于 x64 的处理器
笔和触控	没有可用于此显示器的笔或触控输入

### 问题

问题１：在 PrusaSlicer 启动，配置向导：选取A打印机和B打印机， 下一步。。。 直到选择”耗材配置选区“，切换打印机时，并不能和”耗材配置选区／配置文件＂分别对应。

问题２：在 PrusaSlicer 启动，配置向导：选取A打印机，下一步。。。 直到选择”耗材配置选区“。返回打印机选区，重新选择B打印机，下一步。。。 直到选择”耗材配置选区“，”耗材配置选区／配置文件＂中却显示的是打印机A的默认选项。

### Version 
version_2.4.1-beta3 includes versions prior to 2.4.1.

### System version + hardware information

Version Windows 10 Enterprise
Version number 21H2
Installation Date ‎2021/‎6/‎21
OS Build 19044.1526
Experience Windows Feature Experience Pack 120.2212.4170.0

Device Name CreatBot-APP
Processor Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz 3.70 GHz
On-board RAM 16.0 GB (15.9 GB available)
Device ID 39456730-8DA7-4A13-9689-BDD0D0B270F8
Product ID 00329-00000-00003-AA432
System type 64-bit operating system, x64-based processor
Pen and Touch No pen or touch input available for this monitor

### question

Question 1: When PrusaSlicer starts, the configuration wizard: select A printer and B printer, the next step. . . 　Until "Consumables Configuration Selection" is selected, when switching printers, it cannot correspond to "Consumables Configuration Selection/Profile" respectively.

Question 2: When PrusaSlicer starts, the configuration wizard: select A printer, the next step. . . 　Until "Consumables Configuration Selection" is selected. Return to the printer selection area, re-select B printer, the next step. . . 　Until "Supplies Configuration Selection" is selected, the default options for Printer A are displayed in "Supplies Configuration Selection/Profile".

rtyr, 2022-02-28:
> Question 1: When PrusaSlicer starts, the configuration wizard: select A printer and B printer, the next step. . . 　Until "Consumables Configuration Selection" is selected, when switching printers, it cannot correspond to "Consumables Configuration Selection/Profile" respectively.

Filament/material profiles are likely not compatible with printer B (or filament profile variant for printer B doesn't exist yet). In general, filament/material profiles are compatible only with printers within its vendor space (same printer manufacturer). It can be further restricted by specific printer model, etc. 

> Question 2: When PrusaSlicer starts, the configuration wizard: select A printer, the next step. . . 　Until "Consumables Configuration Selection" is selected. Return to the printer selection area, [...]

Bundled printer have usually some filament profiles marked as "default", these are automatically selected in the wizard.

lukasmatena, 2024-02-19:
Answered, closing.

closed 2024-02-19 by lukasmatena