prusa3d/PrusaSlicer #8002: memory leak in GDI Objects causing lag

hypov8, 2022-02-28:
PrusaSlicer-2.4.0
PrusaSlicer-2.4.1-beta3+win64-202202241030.zip

windows 7 x64

using the layer slider creates lag across all windows apps.

it seems when i scroll the sliced layer height up/down the GDI OBJECTS in windows task manager increase to thousands.

bubnikv, 2022-02-28:
Does the number of GDI object keep growing or does it increase by thousands
and then stabilize?
Is it a GDI resources leak or just an excessive use of GDI resources? The
former would be a killer, but I bet we would likely already find out.
Running out of GDI resources leads to missing controls, controls not
rendered correctly etc.

hypov8, 2022-02-28:
it keeps increasing every time i use the slider.. i can generate about 1000 objects in 15 seconds
not sure max limit but i have seen 6000

hypov8, 2022-03-01:
does this video help?
https://youtu.be/hryH63oyH7E
task manager at bottom

bubnikv, 2022-03-01:
Thanks, that is scary. We will look into it.

hypov8, 2022-03-01:
disabling '**show object height**' seems to fix the issue

[image]

hope this helps

bubnikv, 2022-03-01:
This issue seems to be Windows 7 specific for whatever reason. We have a
hypothesis that we will test. The issue here is that the UI library
wxWidgets shields us from Win32 hiding some implementation details and
quirks of GDI object resource management.

bubnikv, 2022-03-04:
@YuSanka has a fix, however the fix was a little bit risky to be put into the upcoming PrusaSlicer 2.4.1 patch release, we put the fix into PrusaSlicer 2.5.0. 

The resouce leak is real, however it only manifests itself on Windows 7 and it was there in PrusaSlicer 2.3.0 already and it does not seem to be critical if it was reported only once since 2.3.0 release.

Thanks for your help detecting and investigating the issue.

lukasmatena, 2024-02-19:
The fix is in PrusaSlicer since 2.6.0. Closing.

closed 2024-02-19 by lukasmatena