prusa3d/PrusaSlicer #8003: PS Crash after Start

foreachthing, 2022-02-28:
### Version
version_2.5.0-alpha0-107-g1e9951dec

### Operating system type + version
Windows 10 x64

### Behavior
* Start PS
  * After the splash, PS is here but only for a few seconds.
* clear all profiles and restart PS
  * result: 
<details>
  <summary>1. prusa-slicer-console.exe --loglevel 9</summary>

```
C:\Users\<USERNAME>\MyProgramme\PrusaSlicer_a>prusa-slicer-console.exe --loglevel 9
[2022-02-28 07:32:36.159672] [0x00007294] [trace]   Initializing StaticPrintConfigs
[2022-02-28 07:32:36.172638] [0x00007294] [debug]   full path: 7263087213622277716
[2022-02-28 07:32:36.172638] [0x00007294] [debug]   single instance: undefined. other params: prusa-slicer-console.exe;--loglevel;9
[2022-02-28 07:32:36.172638] [0x00007294] [debug]   init wx instance checker 7263087213622277716.lock C:\Users\<USERNAME>\AppData\Roaming\PrusaSlicer-alpha\cache\
[2022-02-28 07:32:36.172638] [0x00007294] [info]    Instance check: Another instance not found or single-instance not set.
[2022-02-28 07:32:36.174634] [0x00007294] [trace]   System language detected (user locales and such): de_CH
[2022-02-28 07:32:36.175629] [0x00007294] [trace]   Best translation language detected (may be different from user locales): en_GB
[2022-02-28 07:32:36.175629] [0x00007294] [trace]   Switching wxLocales to en_GB
[2022-02-28 07:32:36.261400] [0x00007294] [info]    Checking if indices need to be installed from resources...
[2022-02-28 07:32:36.264393] [0x00007294] [info]    Install index from resources: "Anycubic.idx"
[2022-02-28 07:32:36.267387] [0x00007294] [debug]   PresetUpdater: Copying "C:\Users\<USERNAME>\MyProgramme\PrusaSlicer_a\resources\profiles\Anycubic.idx" -> "C:\Users\<USERNAME>\AppData\Roaming\PrusaSlicer-alpha\cache\Anycubic.idx"
[2022-02-28 07:32:36.275362] [0x00007294] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): The system cannot find the file specified
[2022-02-28 07:32:36.285337] [0x00007294] [info]    Install index from resources: "Artillery.idx"
[2022-02-28 07:32:36.287332] [0x00007294] [debug]   PresetUpdater: Copying "C:\Users\<USERNAME>\MyProgramme\PrusaSlicer_a\resources\profiles\Artillery.idx" -> "C:\Users\<USERNAME>\AppData\Roaming\PrusaSlicer-alpha\cache\Artillery.idx"
[2022-02-28 07:32:36.293315] [0x00007294] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): The system cannot find the file specified
[2022-02-28 07:32:36.302291] [0x00007294] [info]    Install index from resources: "BIBO.idx"
[2022-02-28 07:32:36.303288] [0x00007294] [debug]   PresetUpdater: Copying "C:\Users\<USERNAME>\MyProgramme\PrusaSlicer_a\resources\profiles\BIBO.idx" -> "C:\Users\<USERNAME>\AppData\Roaming\PrusaSlicer-alpha\cache\BIBO.idx"
[2022-02-28 07:32:36.309273] [0x00007294] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): The system cannot find the file specified
[2022-02-28 07:32:36.316254] [0x00007294] [info]    Install index from resources: "Creality.idx"
[2022-02-28 07:32:36.319246] [0x00007294] [debug]   PresetUpdater: Copying "C:\Users\<USERNAME>\MyProgramme\PrusaSlicer_a\resources\profiles\Creality.idx" -> "C:\Users\<USERNAME>\AppData\Roaming\PrusaSlicer-alpha\cache\Creality.idx"
[2022-02-28 07:32:36.325229] [0x00007294] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): The system cannot find the file specified
[2022-02-28 07:32:36.333208] [0x00007294] [info]    Install index from resources: "FLSun.idx"
[2022-02-28 07:32:36.335204] [0x00007294] [debug]   PresetUpdater: Copying "C:\Users\<USERNAME>\MyProgramme\PrusaSlicer_a\resources\profiles\FLSun.idx" -> "C:\Users\<USERNAME>\AppData\Roaming\PrusaSlicer-alpha\cache\FLSun.idx"
[2022-02-28 07:32:36.342184] [0x00007294] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): The system cannot find the file specified
[2022-02-28 07:32:36.350163] [0x00007294] [info]    Install index from resources: "gCreate.idx"
[2022-02-28 07:32:36.352157] [0x00007294] [debug]   PresetUpdater: Copying "C:\Users\<USERNAME>\MyProgramme\PrusaSlicer_a\resources\profiles\gCreate.idx" -> "C:\Users\<USERNAME>\AppData\Roaming\PrusaSlicer-alpha\cache\gCreate.idx"
[2022-02-28 07:32:36.359139] [0x00007294] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): The system cannot find the file specified
[2022-02-28 07:32:36.367118] [0x00007294] [info]    Install index from resources: "INAT.idx"
[2022-02-28 07:32:36.369113] [0x00007294] [debug]   PresetUpdater: Copying "C:\Users\<USERNAME>\MyProgramme\PrusaSlicer_a\resources\profiles\INAT.idx" -> "C:\Users\<USERNAME>\AppData\Roaming\PrusaSlicer-alpha\cache\INAT.idx"
[2022-02-28 07:32:36.376093] [0x00007294] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): The system cannot find the file specified
[2022-02-28 07:32:36.382077] [0x00007294] [info]    Install index from resources: "Jubilee.idx"
[2022-02-28 07:32:36.385070] [0x00007294] [debug]   PresetUpdater: Copying "C:\Users\<USERNAME>\MyProgramme\PrusaSlicer_a\resources\profiles\Jubilee.idx" -> "C:\Users\<USERNAME>\AppData\Roaming\PrusaSlicer-alpha\cache\Jubilee.idx"
[2022-02-28 07:32:36.391053] [0x00007294] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): The system cannot find the file specified
[2022-02-28 07:32:36.398035] [0x00007294] [info]    Install index from resources: "LulzBot.idx"
[2022-02-28 07:32:36.400029] [0x00007294] [debug]   PresetUpdater: Copying "C:\Users\<USERNAME>\MyProgramme\PrusaSlicer_a\resources\profiles\LulzBot.idx" -> "C:\Users\<USERNAME>\AppData\Roaming\PrusaSlicer-alpha\cache\LulzBot.idx"
[2022-02-28 07:32:36.407011] [0x00007294] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): The system cannot find the file specified
[2022-02-28 07:32:36.413994] [0x00007294] [info]    Install index from resources: "PrusaResearch.idx"
[2022-02-28 07:32:36.416985] [0x00007294] [debug]   PresetUpdater: Copying "C:\Users\<USERNAME>\MyProgramme\PrusaSlicer_a\resources\profiles\PrusaResearch.idx" -> "C:\Users\<USERNAME>\AppData\Roaming\PrusaSlicer-alpha\cache\PrusaResearch.idx"
[2022-02-28 07:32:36.423966] [0x00007294] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): The system cannot find the file specified
[2022-02-28 07:32:36.435934] [0x00007294] [info]    Install index from resources: "RatRig.idx"
[2022-02-28 07:32:36.437928] [0x00007294] [debug]   PresetUpdater: Copying "C:\Users\<USERNAME>\MyProgramme\PrusaSlicer_a\resources\profiles\RatRig.idx" -> "C:\Users\<USERNAME>\AppData\Roaming\PrusaSlicer-alpha\cache\RatRig.idx"
[2022-02-28 07:32:36.441919] [0x00007294] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): The system cannot find the file specified
[2022-02-28 07:32:36.448899] [0x00007294] [info]    Install index from resources: "TriLAB.idx"
[2022-02-28 07:32:36.452889] [0x00007294] [debug]   PresetUpdater: Copying "C:\Users\<USERNAME>\MyProgramme\PrusaSlicer_a\resources\profiles\TriLAB.idx" -> "C:\Users\<USERNAME>\AppData\Roaming\PrusaSlicer-alpha\cache\TriLAB.idx"
[2022-02-28 07:32:36.457875] [0x00007294] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): The system cannot find the file specified
[2022-02-28 07:32:36.468846] [0x00007294] [info]    Install index from resources: "Ultimaker.idx"
[2022-02-28 07:32:36.470841] [0x00007294] [debug]   PresetUpdater: Copying "C:\Users\<USERNAME>\MyProgramme\PrusaSlicer_a\resources\profiles\Ultimaker.idx" -> "C:\Users\<USERNAME>\AppData\Roaming\PrusaSlicer-alpha\cache\Ultimaker.idx"
[2022-02-28 07:32:36.475827] [0x00007294] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): The system cannot find the file specified
[2022-02-28 07:32:36.482809] [0x00007294] [info]    Install index from resources: "Voron.idx"
[... cut ...]
[2022-02-28 07:41:44.661962] [0x00009284] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): The system cannot find the file specified
[2022-02-28 07:41:44.678917] [0x00009284] [debug]   boost::filesystem::permisions before copy error message (this could be irrelevant message based on file system): The system cannot find the file specified
[2022-02-28 07:41:45.660308] [0x00009284] [debug]   TriangleMesh::repair() started
[2022-02-28 07:41:45.662303] [0x00009284] [debug]   TriangleMesh::repair() finished

C:\Users\<USERNAME>\MyProgramme\PrusaSlicer_a>
```
</details>

So, PS has become unusable at this point ☹️

bubnikv, 2022-02-28:
If you self compiled, then you should be able to run under the debugger and provide us with a stack trace. Would you please do that?

foreachthing, 2022-02-28:
I build PS from command line (on a virtual computer) and the use it on my "local" computer.
The build went just fine.

How would I get the stack trace?

foreachthing, 2022-02-28:
Oooohh... it must be my system ... it kinda works on the vm... I'll be back. In the meantime I close here.

closed 2022-02-28 by foreachthing

foreachthing, 2022-03-01:
Just for the record: I had to rebuild the whole shebang. Now it works again.