prusa3d/PrusaSlicer #8004: build_win.bat -v 16 creates Platform Toolset v143

foreachthing, 2022-02-28:
### Version
version_2.5.0-alpha0-108-g237958819

### Operating system type + version
Windows 10 x64

### Behavior
* Executing `.\build_win.bat -v 16 -s all -d "deps\build\out_deps"`

This will create this line in all deps/build/*.vcxproj files: `<PlatformToolset>v143</PlatformToolset>`. This does not work with VS 2019.
To be able to compile with VS2019, this value should read `v142`.

Output from commmand:
```
PS C:\x> .\build_win.bat -v 16 -s all -d "deps\build\out_deps"
**********************************************************************
** Build Config: RelWithDebInfo
** Build Steps:  all
** Run App:      none
** Deps path:    C:\x\deps\build\out_deps
** Using Microsoft Visual Studio installation found at:
**  C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional
**********************************************************************
** Visual Studio 2019 Developer Command Prompt v16.11.10
** Copyright (c) 2021 Microsoft Corporation
**********************************************************************
Cleaning deps\build ...
Creating C:\x\deps\build\out_deps
-- Building for: Visual Studio 17 2022
-- Selecting Windows SDK version 10.0.19041.0 to target Windows 10.0.19042.
-- The C compiler identification is MSVC 19.31.31104.0
-- The CXX compiler identification is MSVC 19.31.31104.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Program Files/Microsoft Visual Studio/2022/Professional/VC/Tools/MSVC/14.31.31103/bin/Hostx64/x64/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Program Files/Microsoft Visual Studio/2022/Professional/VC/Tools/MSVC/14.31.31103/bin/Hostx64/x64/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- PrusaSlicer deps DESTDIR: C:/x/deps/build/out_deps
-- PrusaSlicer dowload dir for source packages: C:/x/deps/build
-- PrusaSlicer deps debug build: ON
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.35.1.windows.2")
--
Detected 64-bit compiler => building 64-bit deps bundle

-- Deduced boost toolset: msvc-14.3 based on MSVC compiler
-- Generating done
-- Build files have been written to: C:/x/deps/build
Microsoft (R) Build Engine version 16.11.2+f32259642 for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

C:\Program Files (x86)\Microsoft Visual Studio\2019\Professional\MSBuild\Microsoft\VC\v160\Microsoft.CppBuild.targets(439,5): error MSB8020: The build tools for v143 (Platform Toolset = 'v143') cannot be found. To build using the v143 build tools, please install v143 build tools.  Alternatively, you may upgrade to the current Visual Studio tools by selecting the Project menu or right-click the solution, and then selecting "Retarget solution". [C:\x\deps\build\ZERO_CHECK.vcxproj]

ERROR: *** Build process failed at deps step. ***
```
VS 2022 is installed as well, yes. But that should not matter, yes?!
It should still be building for VS 2019 - this is what I expect...

Jebtrix, 2022-03-01:
I have a pull request to fix this https://github.com/prusa3d/PrusaSlicer/pull/7823

foreachthing, 2022-03-01:
@Jebtrix, thanks for your PR. I just tried it and it fixes my issue.
I'll leave this issue open until #7823 gets merged.

amatulic, 2022-03-18:
I don't have VS 2019, only VS 2022 Community Edition. So experimentally I am seeing if it builds with VS 2022. I tried adding the Visual Studio version to the command:
`build_win.bat -d=..\PrusaSlicer-deps -r=console -v=17`

It crunched away for a long time and eventually terminated with the same error as in this report:
`ERROR: *** Build process failed at deps step. ***`

Running the command again resulted in `ERROR: *** Build process failed at app step. ***` - apparently because the Boost package was not found. Is that something that would normally be automatically installed with VS 2019? I recall last year I managed to get SuperSlicer built on Linux to try something and it was a real pain getting all the dependencies straightened out manually. I am wondering if I'm looking at the same sort of effort here with Windows.

lukasmatena, 2022-03-18:
I merged #7823 (thanks @Jebtrix and @jschuh for a review), closing the issue. I hope it works for everybody.

closed 2022-03-18 by lukasmatena

jschuh, 2022-03-18:
@amatulic, Boost doesn't build with VS 2022 (this is known and documented by the Boost team). It looks like VS 2022 support is planned for Boost 1.79 (recently in beta), but even after it's released I expect it will take the Prusa team some time to integrate it.

Everything else in Prusa Slicer should build just fine with vs 2022, however. So, you can just build just the deps with 2019, and then then build the app with 2022 if you want to test it out.

@lukasmatena, thanks, happy to be of help.