prusa3d/PrusaSlicer #8006: Fuzzy Skin

Snow-Eagle, 2022-02-28:
### Version
2.4.0

### Operating system type + version
Windows 11 Pro x64

### 3D printer brand / version + firmware version (if known)
Prusa Mk3+

### Behavior
Fuzzy Skin has the option for; None, Outside Walls, All Walls.
 
Would like to see an option for Top and/or Bottom.
When printing a base for a cartoon character it would be nice to be able to make the top area look like grass.

Alpacad, 2022-03-05:
I don't think this is applicable. _Fuzzy skin_ uses **XY movements** to generate its effect on walls, therefore it can't be used on top/bottom surfaces.
A solution would be to **rotate the base** and having the "top" printed as wall with _fuzzy skin_ activated. If you wish to have a **smooth bottom** on the base, just **use a modifier** and place it where the fuzzy skin is needed.

N.B. if you use this metod and have a bed slinger 3D printer, **print the base along your bed axis** (usually Y) and help yourself with a brim or raft.

[image]

Snow-Eagle, 2022-03-05:
Thanks for your reply. I understand how this works, but my stands are free hand and not square. I was just wandering if the Fuzzy Skin could be added to the Top in the future.

closed 2022-03-08 by Snow-Eagle