prusa3d/PrusaSlicer #8007: [Feature request] Smart fill for enforcing and blocking seam

laptop-laptop, 2022-02-28:
### Version
2.4.1-beta3

_Use `About->About PrusaSlicer` for release versions_
2.4.1-beta3+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
i3MK3S+

### Behavior
* _Describe the problem_

Smart fill in Paint-on supports is very useful. Thank you for making this feature.
Could you please implement this feature for enforcing and blocking seam as well?

_Is this a new feature request?_

Yes

neophyl, 2022-02-28:
Function already exists.  Bottom tool of the ones on the left, seam painting.  Allows you to paint on areas that enforce and block seams.  Shortcut key for it is 'P'

laptop-laptop, 2022-03-01:
I can paint to enforce or block seams manually but I don't see any function to allow the user to choose Tool type to Smart fill like Paint-on support. In Seam painting, I can only choose the Brush size, Brush shape (Sphere, Circle) and Clipping of view.

Snuff1eupagus, 2022-09-19:
I'd like to see "Smart fill" for seam painting as well, that would be very useful. I'd like to see a smaller seam brush size of 0.1 implemented also, like the Paint tool has, the seam tools smallest brush is 0.4 to large for may applications.

closed 2026-09-29 by prusaslicer-bot[bot]