prusa3d/PrusaSlicer #8011: variable layer with supports on

Bergateo, 2022-03-02:
Version of PrusaSlicer used 2.4.1-beta3+win64

windows 10 
64bit 

Flying Bear Ghost 5
Marlin 2.0.x

Hi everyone! has everyone suffered of layer mismatch if the supports are active if supports are active?
If the supports are activated, it seems like the option of variable layer settings but instead it is disabled. If i turn off the supports, the layer height is normal.
[image]
[image]

foreachthing, 2022-03-02:
It's not a mismatch. It's an **additional** 160 layers of support. The height remains the same whereas the layer count goes up.
No issue here ;-)

Bergateo, 2022-03-02:
> It's not a mismatch. It's an **additional** 160 layers of support. The height remains the same whereas the layer count goes up.
> No issue here ;-)

But how it's possible to disable it?

foreachthing, 2022-03-02:
What do you want to disable?
I don't understand what you're trying to report?!

Bergateo, 2022-03-02:
> What do you want to disable?
> I don't understand what you're trying to report?!
I want to disable the 160 layer add that mess up the entire layer height of the model, because it isn't restricted to just support but effect the overall print.

I figured out that if I turn the z distance of the support to a certain value (e.g. 0.1), all the print is set to that 0.1 later even if I want it on 0.2. It's like the height of support affects the all print to a different value. Don't know if this could help in the explanation...

lukasmatena, 2022-03-03:
@Bergateo Support layers are not necessarily synchronized with object layers, which allows slicer to e.g. print thicker support layers and thus reduce print time. If you don't want that, you can check "Synchronize with object layers" in Print Settings->Support material. It should not affect the object though, so I don't understand what problem you are trying to solve. If you believe there is a bug, attach a 3MF and describe it in detail.

lukasmatena, 2022-04-05:
Intended behavior.

closed 2022-04-05 by lukasmatena