prusa3d/PrusaSlicer #8012: [Bug?] Supports Goes Through Another Object

sarf2k4, 2022-03-02:
### Version
Version 2.4.0+win64

### Operating system type + version
PrusaSlicer
Version:   2.4.0+win64
Build:     PrusaSlicer-2.4.0+win64-202112211614

### 3D printer brand / version + firmware version (if known)
Slice preview

### Behavior
Object A has overhang that needs supports, while Object B is small and under the Object A's overhang that needed supports. Slicer will generate a support that goes through Object B.

#### Project File (.3MF) where problem occurs
[Shape-Box.zip](https://github.com/prusa3d/PrusaSlicer/files/8172805/Shape-Box.zip)
[image]
[image]

Wasn't sure if this has been asked before or not, and I'm not sure if; this behavior was intentional, we're not supposed to arrange objects like these or none ever thought of arranging these objects like this. 

One scenario another user perhaps, want to arrange objects to be like this is that they want to print a lot of objects and there are still room/spaces left for the smaller parts to print in one go without having the support goes through another object.

lukasmatena, 2022-03-03:
Duplicate of #316, #1275, #4517. Closing.

closed 2022-03-03 by lukasmatena

fe60, 2022-03-03:
Merging objects solves the problem.
https://github.com/prusa3d/PrusaSlicer/issues/6746#issuecomment-889824207