prusa3d/PrusaSlicer #8013: 2.4.1-Beta 3 - Z Hop Setting Ignored - Causing Failed Prints

thwerks, 2022-03-02:
### Version
Version 2.4.1-beta3+linux-x64-GTK3

### Operating system type + version
Debian 11

### 3D printer brand / version + firmware version (if known)
Ender 3

### Behavior
Z hop is specified but is ignored in various parts of the print.

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
 you can see on layer 14 for example, the layer begins with a travel move over infill lines but does NOT lift Z. print fails repeatedly. 

file renamed from .3mf to .txt
[Hook.txt](https://github.com/prusa3d/PrusaSlicer/files/8172905/Hook.txt)

lukasmatena, 2022-03-03:
@thwerks Could you please post the part of G-Code you are referring to? I don't see the missing zhop.

rtyr, 2022-03-03:
I don't see any issue. "Only retract when crossing perimeters" is enabled. No retract = no z-hop.

thwerks, 2022-03-04:
ah yes, i have made this mistake before. Its easy to overlook that being that its under Infill settings rather than in the retract settings or even perimeter settings. can just close this issue, thank you.

closed 2022-03-04 by FidelCapo