prusa3d/PrusaSlicer #8015: Feature request: custom modifier seam control, dropbox config

eneene, 2022-03-03:
Version: 2.4.0+x64
Build: PrusaSlicer-2.4.0+x64-202112211607
Operating System: Macintosh
System Architecture: 64 bit
System Version: macOS High Sierra Version 10.13.6 (Build 17G13035)

This is a feature request

I have two suggestions.

1-Allowing seam control to be made with custom modifiers would allow us to: first, save the seam positions even if we reload the .stl from disk. Second, it would allow us to control the seam position from our cad software of choice, with much more precision.

2-Allowing for choice on the place we keep the config files (e.g. dropbox) would allow us to have the same profiles for our printers even if we have multiple computers.

Thank you for prusaslicer
#
Thiago.

neophyl, 2022-03-03:
Usually feature requests are done one per issue to make keeping track of them easier.  Its also usual to search if a request already exists.  Such as https://github.com/prusa3d/PrusaSlicer/issues/5023 for your second one for example.

eneene, 2022-03-03:
thanks for the heads up!

dartrax, 2022-03-07:
I agree to your first one, the seam control modifier. I think in (closed) #946 this idea was developed and some related already existing FRs are linked there.

Blayzeing, 2023-09-09:
I'm jumping in here mostly to track this feature request (the first one) because I'd also like it, specifically the ability to place a modifier object that then influences/attempts to limit seam generation to within (or exclude) it.

I'm a C++ programmer, so if I get the time + if it becomes a big enough issue for me, I'll have a closer look and see if I can implement a PR for it. I feel like it'd fit well into the PrusaSlicer ecosystem, UX-wise, but I have my doubts about how do-able it'd be due to seam position currently looking like it's a bit of a separate system from other 3D-based slicing modifiers (what with it being 2D and all). If any devs come across this post and happen to have insight into where I should look or things I should look out for, please do comment.

dartrax, 2023-09-09:
I've tried to implement it myself in a very old version of PrusaSlicer. The PR is here. https://github.com/prusa3d/PrusaSlicer/pull/728
May be it helps you, but I think the code has changed since then...

Blayzeing, 2023-09-09:
Ah that's fascinating @dartrax ! At the very least, it shows how much such a feature is desired. It's a shame it wasn't taken forward alongside the painting approach. But thanks for the wayfinding - I'm sure it'll be useful here! I'm convinced now to have a look and see if I can't at least assess how easy it'll be under the current architecture.

Thanks!

Blayzeing, 2023-09-10:
I've had a poke around in the source code, and I think this'll be do-able, so I'll probably put something together soon.

closed 2026-09-29 by prusaslicer-bot[bot]