prusa3d/PrusaSlicer #8017: Internal error: invalid map<K, T> key

laptop-laptop, 2022-03-03:
### Version
2.4.1-beta 3, rc-1

_Use `About->About PrusaSlicer` for release versions_
2.4.1-rc-1+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
i3MK3S+

### Behavior
* _Describe the problem_

Didn't have such issue with 2.4.0 but with 2.4.1-beta3 and rc-1, if I have PrusaSlicer running on a PC and I connect to it via a remote desktop application from an iPad, I keep getting the follow error and I have to relaunch PrusaSlicer again:

PrusaSlicer Error
Internal error: invalid map<K,T> key
Internal error: invalid map<K,T> key

laptop-laptop, 2022-03-04:
Jump Desktop latest version 9.6.8
I connect using the fluid method

laptop-laptop, 2022-03-10:
I updated to 2.4.1. I noticed another way to cause this problem.

1. Launch PrusaSlicer.
2. Open a project file
3. Switch to another user on the same PC
4. Switch back to the original user
5. 'Unhandled unknown exception; terminating the applications' error shows up and PrusaSlicer crashed and terminated automatically.

Giulio-Machiavelli, 2022-06-23:
I've the same issue; **Internal error: invalid map<K, T> key** on PrusaSlicer 2.4.2+win64.

The error just appeared suddenly while PrusaSlicer was running unused in the background.

mmcglumphy, 2022-06-26:
If my workstation goes to sleep after 10 min, any instance of Slic3r throws this error. Not sure if it happens as the moment of sleep or waking, though. Win 11, nVidia T600.

closed 2026-09-29 by prusaslicer-bot[bot]