prusa3d/PrusaSlicer #8024: Wrong translation (German)

wolfgangmauer, 2022-03-04:
### Version
2.4
2.4.1

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
VORON 2.4

### Behavior
Wrong language translation for inner/outer walls
It is not clear which walls are meant here

[image]
[image]

dartrax, 2022-03-05:
Agreed.
The first item of the legend (yellow box) should be "Kontur", while the second item (orange box) is correct being "Außenkontur".

wolfgangmauer, 2022-03-05:
Maybe its cleaner to call it "Innenkonturen" and "Außenkonturen" ;-)

dartrax, 2022-03-05:
I don‘t think so. The logic of „Außenkontur“ is that it denotes the outer most perimeter. With the same logic, „Innenkontur“ would denote the inner most perimeter. That would be wrong because the middle perimeters also belong to that.

fe60, 2022-03-07:
As stated by @dartrax, translation of "Perimeter" in preview should be changed from "Außenkontur" to "Kontur".

| Usage | English | German | Status | Source |
| --- | --- | --- | --- | --- |
| Preview | Perimeter | Außenkontur | ⚠️ | [PrusaSlicer_de.po#L8731](https://github.com/prusa3d/PrusaSlicer/blob/3bf704932af693c5e47b93ed1811223ab5445b99/resources/localization/de/PrusaSlicer_de.po#L8731) |
| Preview | External perimeter | Außenkontur | ✔️ | [PrusaSlicer_de.po#L4279](https://github.com/prusa3d/PrusaSlicer/blob/3bf704932af693c5e47b93ed1811223ab5445b99/resources/localization/de/PrusaSlicer_de.po#L4297) |
| Config | Perimeters | Konturen | ✔️ | [PrusaSlicer_de.po#L8740](https://github.com/prusa3d/PrusaSlicer/blob/3bf704932af693c5e47b93ed1811223ab5445b99/resources/localization/de/PrusaSlicer_de.po#L8740) |
| Config | External perimeters | Außenkonturen  | ✔️ | [PrusaSlicer_de.po#L4302](https://github.com/prusa3d/PrusaSlicer/blob/3bf704932af693c5e47b93ed1811223ab5445b99/resources/localization/de/PrusaSlicer_de.po#L4302) |

lukasmatena, 2022-04-05:
This should be fixed in the just released 2.4.2-rc1. Closing.

closed 2022-04-05 by lukasmatena