prusa3d/PrusaSlicer #8025: Multi color infill mostly solid due to different colors

Blazenlink, 2022-03-04:
### Version
2.4.0 Release
2.4.1-rc1

### Operating system type + version
Windows 10

### Behavior
The whole object I used multicolor painting in is mostly 100% infill due to the color change lines going through the center of the object. 

[image]

SomeoneElse58, 2022-03-05:
You can use the **Maximum width of a segmented region** setting in PrintSettings>MultipleExtruders>Advanced to [set the depth the colors go into the print from the perimeter](https://github.com/prusa3d/PrusaSlicer/issues/6892#issuecomment-914181121).

Blazenlink, 2022-03-05:
Awesome that worked, Thank you.

closed 2022-03-05 by Blazenlink