prusa3d/PrusaSlicer #8026: MacOS - Unable to manually click on scroll bars and drag position.

autonumous, 2022-03-04:
### Version
Many, eg
2.4
2.4.1beta
2.4.1rc1

### Operating system type + version
macOS 11.6.4 (Big Sur)

### Behavior
When selecting a list that contains a number of items, eg Filament list, a scroll bar is shown.  You should be abel to select this and drag with mouse.   In PrusaSlicer, this scroll bar appears, but I'm unable to drag.  Can only scroll with mouse wheel/trackpad 

This issue has existed for as long as I remember.

dartrax, 2022-03-15:
Also reported in #7874 (which is nearly impossible to find due to screenshots only and meaningless title)

autonumous, 2023-09-27:
Any fix for this.  Really annoying with large numbers of entries in menu..

closed 2026-09-29 by prusaslicer-bot[bot]