prusa3d/PrusaSlicer #8027: Cant open the sclicer.

scooter024, 2022-03-05:
I tried to open the slicer and I got this pop up. I deleted the slicer off my computer and downloaded again and still got the same thing. I tried to follow the folders listed but I get nothing. I'm not the best woth computers so idk what to do. 
[image]

dartrax, 2022-03-05:
Your image is not visible, could you try uploading it again and wait until the upload finished before submitting?

scooter024, 2022-03-05:
> Your image is not visible, could you try uploading it again and wait until the upload finished before submitting?

Ok I've never used thus site before I'm not a tech type of guy. Is it pk to post my issue on this site or is it only for code stuff? Also the pic should be visible now.

dartrax, 2022-03-05:
Thanks for uploading the picture, you are perfectly right on this site for that kind of stuff!

> I tried to follow the folders listed but I get nothing. 

Did you enable viewing of hidden files in windows explorer? The default is that the Appdata folder is hidden. You can type %appdata% in the address bar of windows explorer to get into that folder (I believe that works even without enabling hidden files).
You can rename the folder %appdata%\PrusaSlicer to %appdata%\PrusaSlicer_Backup and try to start the slicer again.
Please keep the backup in case developers want to take a look into that to understand why this happened.

scooter024, 2022-03-05:
Windows explorer are you referring to internet explorer?

dartrax, 2022-03-05:
No, your file browser.

scooter024, 2022-03-05:
Ohh ok. I'll check when I get home.

scooter024, 2022-03-06:
I found the settings to show hidden folder and I was able to follow the
files to the "my printer" file. I tried to rename it like u said but it
would o a error and wouldn't work. I then moved the my printer file to a
empty folder on my desktop and now the slicer will open and work

bubnikv, 2022-03-08:
great. I believe this issue is resolved. 
Closing

closed 2022-03-08 by bubnikv