prusa3d/PrusaSlicer #8028: Too little room for +/- buttons in GTK3 version

Area5142, 2022-03-05:
### Version
_PrusaSlicer-2.4.1-rc1+linux-x64-GTK3-202203031441_

### Operating system type + version
_Linux Mint 19.3 x64 Cinnamon_

### 3D printer brand / version + firmware version (if known)
_All printers_

### Behavior
* _Describe the problem_
* Too little room for the + button in dialogs with +/- value spin buttons:
[image]

This is consistent over tab dialogs, not only the shown example.

The GTK3 version has the problem - compiling to native GUI do not have any problems on Linux. I normally avoid using the GTK2/GTK3 versions and use the native version, but that was not offered in this RC-1 release.
Compiling the master code gives me version PrusaSlicer-version_2.5.0-alpha0-118-g3bf704932 instead of PrusaSlicer 2.4.1-rc1.

* _Expected Results_
Both + and - button displayed.

* _Actual Results_
The + button is cut in half.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
Not needed.

lunatic1972, 2022-03-11:
I've had similar issues in the past and opened a request but I don't believe they ever got to it. I'm assuming GTK3 doesn't have the same resize capabilities like windows API's but GTK2 had a lot of workarounds for doing things

Area5142, 2022-03-11:
>I'm assuming GTK3 doesn't have the same resize capabilities like windows API's but GTK2 had a lot of workarounds for doing things

GTK2 version works as expected, it uses another graphical spin element, not buttons.

lukasmatena, 2024-02-19:
Correct me if I am wrong, but I believe that this is fixed as of 2.7.2-alpha2. Closing.

closed 2024-02-19 by lukasmatena