prusa3d/PrusaSlicer #8031: [Feature request] Obtaining and reaching the layer height pointed by the mouse pointer

laptop-laptop, 2022-03-05:
### Version
2.4.1-rc1

_Use `About->About PrusaSlicer` for release versions_
2.4.1-rc1+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
i3MK3S+

### Behavior

When using height range modifier, we need to use the arrows on the keyboard and feedback from the visualization to find a specific layer number that we want to do some modifications. It would be nice if we could just click on the surface in the visualization using the mouse and PS automatically moves the slider pointer to that layer.

_Is this a new feature request?_

Yes

dartrax, 2022-03-15:
Could be part of #7551.

dartrax, 2026-09-18:
Still a valid FR