prusa3d/PrusaSlicer #8032: Vase mode Preview not working anymore

how-eee, 2022-03-05:
Version: 2.4.1-rc1+win64
Build: PrusaSlicer-2.4.1-rc1+win64-202203031432
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 18363), 64-bit edition

### Behavior
Gcode preview shows nothing for sliced in vase mode model

how-eee, 2022-03-05:
[image]

rtyr, 2022-03-07:
Please provide your 3MF project file.

how-eee, 2022-03-11:
[vase.zip](https://github.com/prusa3d/PrusaSlicer/files/8231858/vase.zip)
 Last relised 2.4.1 has same issue.
Just Add primitive shape - Box. Slice  it in vase mode.
Nothing showed in preview.

neophyl, 2022-03-11:
Seems to be related to the gcode flavour.  Some show in the preview and other dont.  For example Teacup or Sailfish also doesnt display anything like Reprap/Sprinter but plain reprap does.  Using 2.4.1.

rtyr, 2022-03-11:
Thanks. I was able to reproduce it in 2.4.1, not in 2.4.0.

It seems to happen when there is no Z travel in the start g-code. You can add for example `G1 Z0.2 F1500`. It should fix the issue for now. We will look into it.

lukasmatena, 2022-04-05:
This should be fixed in the just release 2.4.2-rc1. Can you please confirm? Thanks.

PeterH1500, 2022-04-07:
2.4.2rc1 fixes the bug that hides the first layer in vase mode, but leaves behind a bug (or part of a bug), present in 2.4.1, that hides one wall of a square/rectangular vase in all layers except the first. The -Y wall is missing - see screen capture..
[image]
.

rtyr, 2022-04-07:
@PeterH1500 Could you please provide your 3MF project?

PeterH1500, 2022-04-08:
No problem (apart from github arguing about file types!) I've changed the filename extension from .3mf to .txt (strangely, github wouldn't accept a zip file.)

[PLA 100mm Cube.txt](https://github.com/prusa3d/PrusaSlicer/files/8452935/PLA.100mm.Cube.txt)

rtyr, 2022-04-08:
Thanks. I was not able to reproduce the issue with your 3MF. It is OK on my PC.

how-eee, 2022-04-10:
now it looks like fixed/ Thanks

closed 2022-04-10 by how-eee