prusa3d/PrusaSlicer #8033: macOS Monterey external display crash

mitsumaui, 2022-03-06:
### Version
2.4.0+arm64 & 2.4.1-rc1

### Operating system type + version
Mac OS X Monterey 12.2.1

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
Crash on opening whilst using external display from 2020 MBP M1
Crash when moving window from internal display to external. 
No problem when program opened / used on built-in display. 

`Process:               PrusaSlicer [1246]
Path:                  /Applications/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.4.0+x64-202112211607 (PrusaSlicer-2.4.0+x64-202112211607)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-03-06 08:46:05.1226 +0000
OS Version:            macOS 12.2.1 (21D62)
Report Version:        12
Anonymous UUID:        BE32A704-0424-4324-951F-17E9F574835F

Sleep/Wake UUID:       DA8B4F9C-691C-4027-B387-D97772008976

Time Awake Since Boot: 1000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called

Thread 0 Crashed:: slic3r_main Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x198bc59b8 __pthread_kill + 8
1   libsystem_pthread.dylib       	       0x198bf8eb0 pthread_kill + 288
2   libsystem_c.dylib             	       0x198b36314 abort + 164
3   PrusaSlicer                   	       0x1039fd77c wxAbort() + 12
4   PrusaSlicer                   	       0x103a92a30 wxEvtHandler::WXConsumeException() + 192
5   PrusaSlicer                   	       0x103a92944 wxEvtHandler::SafelyProcessEvent(wxEvent&) + 44
6   PrusaSlicer                   	       0x1039676e0 wxWindow::MacDoRedraw(long) + 708
7   PrusaSlicer                   	       0x1039e2908 wxWidgetCocoaImpl::drawRect(void*, NSView*, void*) + 672
8   PrusaSlicer                   	       0x1039e0ab4 wxOSX_drawRect(NSView*, objc_selector*, CGRect) + 132
9   AppKit                        	       0x19bee12d4 -[_NSOpenGLViewBackingLayer display] + 960
10  QuartzCore                    	       0x19fb26c70 CA::Layer::display_if_needed(CA::Transaction*) + 776
11  QuartzCore                    	       0x19fc6939c CA::Context::commit_transaction(CA::Transaction*, double, double*) + 528
12  QuartzCore                    	       0x19fb08ff0 CA::Transaction::commit() + 680
13  AppKit                        	       0x19b98454c __62+[CATransaction(NSCATransaction) NS_setFlushesWithDisplayLink]_block_invoke + 304
14  AppKit                        	       0x19c0e64ac ___NSRunLoopObserverCreateWithHandler_block_invoke + 64
15  CoreFoundation                	       0x198cc3e4c __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 36
16  CoreFoundation                	       0x198cc3c98 __CFRunLoopDoObservers + 572
17  CoreFoundation                	       0x198cc31d0 __CFRunLoopRun + 764
18  CoreFoundation                	       0x198cc2734 CFRunLoopRunSpecific + 600
19  HIToolbox                     	       0x1a185af68 RunCurrentEventLoopInMode + 292
20  HIToolbox                     	       0x1a185abf4 ReceiveNextEventCommon + 320
21  HIToolbox                     	       0x1a185aa9c _BlockUntilNextEventMatchingListInModeWithFilter + 72
22  AppKit                        	       0x19b819000 _DPSNextEvent + 844
23  AppKit                        	       0x19b8178a4 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1332
24  AppKit                        	       0x19b8099b4 -[NSApplication run] + 596
25  PrusaSlicer                   	       0x1039c792c wxGUIEventLoop::OSXDoRun() + 160
26  PrusaSlicer                   	       0x103a9afac wxCFEventLoop::DoRun() + 40
[... 126 lines cut ...]
AirPort: Wi-Fi, wl0: Oct 26 2021 16:10:41 version 18.20.310.15.7.8.120 FWID 01-5d6c3867
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Ethernet Adapter (en5), Ethernet, en5
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: HP USB-C Dock G5
USB Device: USB 10/100/1000 LAN
USB Device: USB5734
USB Device: 4-Port USB 3.1 Hub
USB Device: HP USB-C Dock G5
USB Device: HP USB-C Dock G5
USB Device: USB2734
USB Device: 4-Port USB 2.1 Hub
USB Device: SMSL AD18 AMP
USB Device: Jabra SPEAK 410 USB
USB Device: Microsoft Pro Intellimouse
USB Device: Drevo keyboard
USB Device: USB Audio
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
`

#### Project File (.3MF) where problem occurs
N/A

mitsumaui, 2022-04-15:
Issue still present in 2.4.2-rc2

lukasmatena, 2023-03-27:
Is it still happening in 2.6.0-alpha5? Thanks.

closed 2026-09-29 by prusaslicer-bot[bot]