prusa3d/PrusaSlicer #8038: Feature Request:  Thumbnails, build plate image remove

K2Kevin, 2022-03-07:
Would like the feature to remove background image (Buildplate) to thumbnails.  Feature offered in Superslicer which really needs to be in PrusaSlicer.  It makes such a difference in Octoprint.

K2Kevin, 2022-07-25:
> Would like the feature to remove background image (Buildplate) to thumbnails. Feature offered in Superslicer which really needs to be in PrusaSlicer. It makes such a difference in Octoprint.

still not interested in this?

spicewiesel, 2022-10-13:
This is just a cosmetic things, yes, but it would look so much nicer in the UI of Octoprint and Mainsail:

[image]
[image]

mastercartman, 2023-08-18:
Same for me! I use to print in black ASA and the preview in octoprint and Fluidd is really hard to see with the dark printbed behind a black part. I know, I could set the color to a standard orange or anything, but I like seeing the "real world" preview.

AlexMatter1512, 2023-10-25:
I really hope this feature will actually be added ad some point...

Jan-Soustruznik, 2024-02-22:
SPE-1715

anarsoul, 2024-05-14:
It's pretty trivial to remove the bed from thumbnail, see https://gist.github.com/anarsoul/ffcf57d49ee7bddb0c8b6a311b2a7464

I don't really see much point in having print bed in the thumbnail and no other slicers have it. I can open a merge request with the change above if the devs are OK with just removing it without having an config option to add it back.

Terror-Gene, 2024-08-23:
This is the biggest issue i have with PrusaSlicer.
Love everything else about it, except it makes these ugly thumbnails.
been putting up with it for ages thinking "surely they will add the option soon"... years passed.... still waiting.
I loved the transparent build plate & thumbnail color options in SuperSlicer, makes klipperscreen & fluidd look so much nicer!

ramosglauco, 2024-10-14:
Agree!!! BIG W

Kaelum, 2025-05-16:
@Jan-Soustruznik It's not just a cosmetic thing, as the thumbnails actually cause the touch screen on Creality printers to lockup.

closed 2026-09-29 by prusaslicer-bot[bot]