prusa3d/PrusaSlicer #8039: Prusa Filament selection missing

wisco-kid, 2022-03-07:
### Version
_Version of PrusaSlicer used goes here_ 2.4.1rc

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_ 64bit
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known) Prusa Mini+
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_ Prusa

### Behavior
* _Describe the problem_ Can't select any factory filament selection.
* _Steps needed to reproduce the problem_ unknown
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_ no profiles for filament from Prusa
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

lukasmatena, 2022-03-09:
Please, provide more information.

ChasonDeshotel, 2022-03-27:
I have this issue on SuperSlicer--is there anything useful I can contribute to this post from SuperSlicer or are there a set of steps I can follow to reproduce this in Prusa?

I know this is an issue but I'm not sure how I got into this state

There's a bit more info and a screenshot in the linked case

rtyr, 2022-03-28:
There is not enough information. Filament profiles from our Prusa Research bundle are visible only for Prusa3D printer profiles or for printer profiles inherited from them.

ChasonDeshotel, 2022-03-28:
On #2001 it's stated "this is the firmware repository, not the Slic3r repository. They are separate. The Slic3r PE repository is here: https://github.com/prusa3d/Slic3r/issues" which prompted me to find a matching case here. But I've just noticed that the link redirects here... were these merged?

Can I just add additional information to #2001?

rtyr, 2022-03-28:
https://github.com/prusa3d/PrusaSlicer/issues/2001 was originally opened in Firmware repository (by mistake). Later the issue was manually transferred to correct repository. The issue is already closed.

You can archive your data folder (help-show configuration folder) to zip and attach it here with the information, which profiles should be visible in your opinion.

rtyr, 2022-05-09:
No response. Closing.

closed 2022-05-09 by rtyr