prusa3d/PrusaSlicer #8040: Position of head after print

livius2, 2022-03-07:
### Version
2.4.0

### Behavior
Currently after print head go to the max left. My position of printer cause it will be better if it go after print to max Right.
Add option (checkbox) where we can simply click on it.

Currently Head after print is too low for me if i print low profile objects.
Please add option to specifi min Z-Position after print (EditBox).

_Is this a new feature request?_
**Yes**

foreachthing, 2022-03-08:
Edit your End G-Code section to do just what _you_ want.
Below is just an example, I did _not_ test it (use at own risk):

```
; # # # # # # START Footer
G91 				; relative!!! coordinates
G1 Z+5  X-10 Y-10 E-8 F15000	; remove head and retract
G0 X{print_bed_max[0]} F15000	; go max (bed size) right
; G28 X0 Y0			; move X/Y to min endstops
M104 S0				; extruder heater off
M140 S0				; heated bed heater off (if you have it)
M84 				; disable motors
; # # # # # # END Footer
```

livius2, 2022-03-08:
Even if your answer is correct(i do not know that) consider that i can be a user which use not **Expert** or even **Advanced** tab only **Simple**. He/she should be able to do this in simple way.

And about **z-axis** positioning after print. I do not want that head go always to such configured position, only i need to setup **minimal z-axis position**. E.g. if i set Z-axis min position to 50mm, and my print is 10mm height it should go to position 50, but if my print is higher e.g 70mm i need that printer head go higher then 70mm and do not go down to 50mm ;-)

foreachthing, 2022-03-08:
Well, this is simple, in a way: you set it once and your done. You can even transfer the settings file to another computer or just copy/paste certain parts.

Z-Positioning: If you use `G91`, it will use _relative_ positioning. Example: your print is 51.3 mm tall, then `G1 Z10` will move to Z=61.3 mm. That way you won't run into the problem of forgetting to set the correct height and avoid a crash.
The limit is your printer size. If you print 2 mm below max height and want to move Z+10 then ... I don't know what will happen :-o.

HTH

livius2, 2022-03-08:
As you can see you even do not know how to do this correctly.
If i add gcode then it will not be related to current situation with the model printed.
Only slicer can generate proper code here. After slicing it know what was the max Z position and it can decide to move +10, +20, or ++++.
If after slicing max position was 23 mm and i will have property to set min z-height e.g. 60mm then it can add automaticaly move Z-axis by 60-23=37mm, simple for the slicer, hard for us.

foreachthing, 2022-03-08:
See here for placeholders: https://help.prusa3d.com/en/article/list-of-placeholders_205643

neophyl, 2022-03-08:
If you add a default mk3 printer and examine its end gcode, Prusa use the following line entered into the End gcode section of the printer definition
{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+30, max_print_height)}{endif} ; Move print head up

This moves the print head up 30 or to the maximum height as defined by the printers max height.  If you wanted it higher you would just change 30 to a bigger number.

Just add the same into your printer Settings end gcode section and save the profile.  The use that printer profile to slice and it will do it from then on.

livius2, 2022-03-08:
Isn't it enought reason to have it configurable by simply Checkbox and EditTextBox?
Consider users using "Simple" mode. Why i need to know gcode at all here?
This is something like telling someone which work with MS Word, you can use macro in VisualBasic, it is easy. Yes, it is easy if someone know it.

foreachthing, 2022-03-08:
@neophyl why is `max_print_height` not listed on the placeholder website...? I was looking for that :raised_eyebrow:

@livius2 sometimes you just have to do with what you have and, like with Word and Excel, dig a bit deeper. 
I don't think that PS will change its free software - Also, neophyl has already provided your solution.

>  Yes, it is easy if someone know it.

Well, risking a thumbs down or cake-in-my-face, this might be the time to "dig deeper" and get involved with programming ;-)

livius2, 2022-03-08:
>  "dig deeper" and get involved with programming ;-)

You mean to create pull request?

neophyl, 2022-03-08:
> @neophyl why is `max_print_height` not listed on the placeholder website...? I was looking for that 🤨

You would have to take that up with Prusa.  As its used in their own end gcode I'm a bit surprised its missing from the knowledge base article.  I think it used to be in the list when it was part of the wiki on here before it was moved to the Knowledge base.
While the list is not quite the same due to the divergence, the one for Super Slicer located here https://github.com/supermerill/SuperSlicer/wiki/Macro-&-Variable-list seems to be much more comprehensive.

@livius2 ivius2, I suspect @foreachthing Foreachthing means learning the basics of gcode, which isn't really programming in the classic sense.

foreachthing, 2022-03-08:
> You mean to create pull request?

@livius2, well, if you think you're up to it, go for it!! _(I don't know any c++, so this is out of my league)_
But for now, I think, G-Code will suffice ;-)

livius2, 2022-03-08:
@foreachthing
From your comments it looks like there is no interest and general demand to have such options? 
Is this really rare demand and i only need it? :)

livius2, 2022-03-08:
>> I pretty much agree with @foreachthing , anyone who wants to control the way their printer behaves will end up figuring out the basics of gcode

then why there are options in the slicer at all? ;-) You do not need to provide all options in gcode, you have checkboxes, edits, spinedits ...
This is only another option on some printer page in the prusa slicer

neophyl, 2022-03-08:
Hey you asked for an opinion on if only you wanted it :)  
If Prusa want to add it in as a setting then great.  Its just such an easy thing to do with only a modicum of knowledge though.

The thing is that I don't think there are any other settings that directly add anything into your start and end gcode like that at the moment.  The other settings you mention control how the objects are sliced.  Not what happens before and after.  
If they added a setting in then that would likely involve much effort to code as it would be new functionality.  Beyond that it would have to make sure not to conflict with any other end gcode that someone may have added, so that's further complex checks as the point of the custom gcode areas is to allow the flexibility for users to do things their own way and there could be all sorts of stuff in there for frankly almost unimaginable use cases.

Those sections all taken care of in the printer profiles when you add a printer.   If you are adding a custom printer then you are assumed to know what you are doing.  Otherwise you have to use the ones included, which for non Prusa printers are provided by the community so reliant on how they have theirs set up.

livius2, 2022-03-08:
I wrongly assumed that you both are from PrusaSlicer developers ;-) I am new here.
Now i understand more your comments :)

rtyr, 2022-03-10:
I believe it was answered. It's already possible to customize end g-code and create custom printer profile with different parking position, etc. There is almost infinite number of various possible variants of the end g-code.

I don't know which printer do you have, but I believe we are using similar parking position to the firmware parking position (for example when you stop the print).

Closing.

closed 2022-03-10 by rtyr