prusa3d/PrusaSlicer #8048: How does Height Range Modifier and Variable Layer Height work together?

laptop-laptop, 2022-03-10:
### Version
2.4.1

_Use `About->About PrusaSlicer` for release versions_
2.4.1+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
i3MK3S+

### Behavior
* _Describe the problem_

My default layer height is 0.15mm. I used Height Range Modifier to modify some settings at several layers. Then, used Variable Layer Height to make the surface smooth. However, some layer height became less than or more than 0.15mm. Am I supposed to just leave the small changes as is, go back to Height Range Modifier to change the 'Start at height', 'Stop at height' and 'Layer height' which may not accept 0.3mm or apply Variable Layer Height first. Then, do a Height Range Modifier? What is the proper workflow if I want to use both functions?

rtyr, 2022-03-10:
> My default layer height is 0.15mm. I used Height Range Modifier to modify some settings at several layers. Then, used Variable Layer Height to make the surface smooth. However, some layer height [...]

I am not sure what exactly do you want to achieve with that height range modifier, but I would set variable height first and then set height range modifier according to final layer heights. It might not be super intuitive at the moment, but shouldn't be a problem. I don't know what do you exactly mean by 0.3mm, but if you need wider range than default 0.07-0.25, you can modify the range in printer settings->extruder. Variable layer height feature overrides layer height set in the HR modifier.

> Also, if I want the smoother printing, under Variable layer height, shall I move the Adaptive and Smooth sliders to the right (i.e. 1 for Adaptive and 10 for Radius)?

See https://help.prusa3d.com/en/article/variable-layer-height-function_1750#setting-up-variable-layer-height or contact our tech support.

laptop-laptop, 2022-03-10:
Thank you. Currently, I am using the default 'Layer height limits' of 0.07-0.25. After setting Variable Layer Height, I hit the "Slice now" button. Then, the vertical slider shows 0.2 (1). Next step up is 0.31 (2). I guess I need to change the Max Layer height limit to 0.31 or whatever largest layer height the Variable Layer Height function created. When using Variable Layer Height, what is the recommended layer height for the first layer?

I have read the article you mentioned. It is confusing. For example, under Quality/Speed, it says 'By moving this slider you can change the ratio of small and tall layers. but after Adaptive, it is labeled as Quality/Speed. I would assume moving the slider to the right means largest quality to speed ratio which is the best quality. However, one support agent said moving the slider to the left means the highest quality. As some support staff don't know PrusaSlicer well, I am asking the developers here.

laptop-laptop, 2022-03-24:
Can anybody who knows Quality/Speed and Smooth well clarify my questions please?

rtyr, 2022-03-24:
You should immediately see layer height curve in the preview, so it should be quite self-explanatory. Left side = quality, right side = print time ("speed"). If you move slider closer to the quality side (left), you will print with lower layer heights in general (which takes more time). "Speed" side will create more significant layer height changes based on the geometry of the model. You can smooth those layer height changes by the smooth button, it will make transitions between different layer heights more gradual, so it will be less visible on the print. It is literally up to the user if or how many times to use the smooth button. It can be different for each specific print/model. I would say it will be usually between 1-3, but it really depends on the decision of the user based on the preview.

Closing. Please contact our tech support with further questions.

closed 2022-03-24 by rtyr