prusa3d/PrusaSlicer #8054: (feature request) setting for angles between infill layer

Sandiegoo, 2022-03-11:
Is it possible to set the angle between the infill of 2 Layers. At the moment it is always 90°.
For some Parts it is necessary to have the strains in a certain direction. Especially for Parts where there is a load on it.

 I couldn´t find any settings for it in the prusa slicer.

Can somebody help me with this?

lukasmatena, 2022-03-14:
It is not possible. You can rotate the infill as a whole with respect to the object (Print Settings->Infill->Fill angle), but the angle between consecutive layers is defined by the selected infill type and cannot be customized. It does not even make sense for many infill types, such as gyroid etc). There are many infill types to choose from, which should be enough to get the properties you want.

Sandiegoo, 2022-03-14:
Is it possible to add this setting for the next version?
I see that it does not make sense for the infill pattern. But it would make sense for the solid infill.

dartrax, 2022-03-15:
fyi: I also used that in the past with orher slicers which support that. Namely Simplify3D and IdeaMaker.

yahbluez, 2023-02-27:
I second that feature request.

For 100% infills, like you need to print really transparent, it is necessary to print every layer of infill in the same direction.

@lukasmatena  
Your opinion that this is impossibly is wrong.
Other slicers already can do that for example: cura, simplyf3d, IdeaMaker.
While you are right that this makes no sense for gyroid as example it makes sense for other uses of other infills.

espr14, 2025-01-23:
That would be great. One would use angle of 137.5° (golden ratio angle). This will never repeat the same angle again so it decreases chance that the plate would crack in a specific angle.

Tupson444, 2026-09-14:
still relevant