prusa3d/PrusaSlicer #8056: Crashes when editing Filament Profiles Selection on macOS

rockydevnet, 2022-03-12:
### Version
PrusaSlicer-2.4.0+x64

### Operating system type + version
macOS Big Sur
Version 11.6.2

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S+

### Behavior
* When attempting to add another filament to the Filament Profiles Selection, crashes when clicking Finish in the Configuration Assistant
#### Steps to reproduce
* Initially, PrusaSlicer Configuration Assistant has only been configured with Prusament PLA in the Filament Profiles
* Open PrusaSlicer
* In Filament dropdown, select `Add/Remove filaments` to open the Configuration Assistant
* Select PETG in the type box, and then check the box next to Prusament PETG in the Profile box
* Click Finish

##### Expected:
Prusament PETG profile should have been added to the available filaments

##### Actual
Crashes

Stacktrace from Apple Crash Report:
```

Thread 0 Crashed:: slic3r_main  Dispatch queue: com.apple.main-thread
0   com.prusa3d.slic3r/           	0x000000010a17eafd bool std::__1::operator!=<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::less<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::allocator<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > >(std::__1::map<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::less<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::allocator<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > > const&, std::__1::map<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::less<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::allocator<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > > const&) + 45
1   com.prusa3d.slic3r/           	0x000000010a17ddd5 Slic3r::GUI::ConfigWizard::priv::apply_config(Slic3r::AppConfig*, Slic3r::PresetBundle*, Slic3r::PresetUpdater const*, bool&) + 7685
2   com.prusa3d.slic3r/           	0x000000010a181461 Slic3r::GUI::ConfigWizard::run(Slic3r::GUI::ConfigWizard::RunReason, Slic3r::GUI::ConfigWizard::StartPage) + 609
3   com.prusa3d.slic3r/           	0x0000000109f6b0a9 Slic3r::GUI::GUI_App::run_wizard(Slic3r::GUI::ConfigWizard::RunReason, Slic3r::GUI::ConfigWizard::StartPage) + 153
4   com.prusa3d.slic3r/           	0x000000010acfdb1e wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 222
5   com.prusa3d.slic3r/           	0x000000010acfd88b wxEvtHandler::ProcessEventLocally(wxEvent&) + 59
6   com.prusa3d.slic3r/           	0x000000010acfd7a0 wxEvtHandler::ProcessEvent(wxEvent&) + 96
7   com.prusa3d.slic3r/           	0x000000010acfdcbc wxEvtHandler::SafelyProcessEvent(wxEvent&) + 12
8   com.prusa3d.slic3r/           	0x000000010ab07b22 wxMenuBase::DoProcessEvent(wxMenuBase*, wxEvent&, wxWindow*) + 130
9   com.prusa3d.slic3r/           	0x000000010ab07a04 wxMenuBase::SendEvent(int, int) + 196
10  com.prusa3d.slic3r/           	0x000000010abcd181 wxMenu::HandleCommandProcess(wxMenuItem*, wxWindow*) + 129
11  com.apple.AppKit              	0x00007fff230feebb -[NSApplication(NSResponder) sendAction:to:from:] + 288
12  com.apple.AppKit              	0x00007fff232018c9 -[NSMenuItem _corePerformAction] + 413
13  com.apple.AppKit              	0x00007fff232015e8 -[NSCarbonMenuImpl performActionWithHighlightingForItemAtIndex:] + 95
14  com.apple.AppKit              	0x00007fff2324dc38 -[NSMenu performActionForItemAtIndex:] + 113
15  com.apple.AppKit              	0x00007fff2324dbbe -[NSMenu _internalPerformActionForItemAtIndex:] + 82
16  com.apple.AppKit              	0x00007fff2324da05 -[NSCarbonMenuImpl _carbonCommandProcessEvent:handlerCallRef:] + 101
17  com.apple.AppKit              	0x00007fff231e41d1 NSSLMMenuEventHandler + 1078
```

rockydevnet, 2022-03-12:
Full crash report:

[PrusaSlicerCrashReport.txt](https://github.com/prusa3d/PrusaSlicer/files/8237300/PrusaSlicerCrashReport.txt)

lukasmatena, 2022-03-12:
Can you please try in 2.4.1? Similar problem was already fixed (#7655), it is possible this is the same.

rockydevnet, 2022-03-12:
Upgrading to 2.4.1 appears to have resolve the issue, many thanks @lukasmatena

closed 2022-03-12 by rockydevnet