prusa3d/PrusaSlicer #8058: Upgrade to 2.4.1 causes slowdown

msproul, 2022-03-12:
### Version
Version 2.4.1+linux-x64-GTK3

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Linux  5.13.0-35-generic #40~20.04.1-Ubuntu SMP Mon Mar 7 09:18:32 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
PrusaSlicer
Version:   2.4.1+linux-x64-GTK3
Build:     PrusaSlicer-2.4.1+linux-x64-GTK3-202203101056

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 5.13.0-35-generic x86_64
Total RAM size [MB]: 67,328MB
OpenGL installation
GL version:   3.1 Mesa 21.2.6
Vendor:       Mesa/X.org
Renderer:     llvmpipe (LLVM 12.0.0, 256 bits)
GLSL version: 1.40
<details>
<summary>Installed extensions:</summary>
GL_3DFX_texture_compression_FXT1
GL_AMD_conservative_depth
GL_AMD_draw_buffers_blend
GL_AMD_multi_draw_indirect
GL_AMD_query_buffer_object
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_stencil_export
GL_AMD_shader_trinary_minmax
GL_AMD_texture_texture4
GL_ANGLE_texture_compression_dxt3
GL_ANGLE_texture_compression_dxt5
GL_APPLE_packed_pixels
GL_ARB_ES2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gl_spirv
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_indirect_parameters
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multi_draw_indirect
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_parallel_shader_compile
GL_ARB_pipeline_statistics_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_polygon_offset_clamp
GL_ARB_post_depth_coverage
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_query_buffer_object
GL_ARB_robust_buffer_access_behavior
GL_ARB_robustness
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_seamless_cubemap_per_texture
GL_ARB_separate_shader_objects
GL_ARB_shader_atomic_counter_ops
GL_ARB_shader_atomic_counters
GL_ARB_shader_bit_encoding
GL_ARB_shader_draw_parameters
GL_ARB_shader_group_vote
GL_ARB_shader_image_load_store
GL_ARB_shader_image_size
GL_ARB_shader_objects
GL_ARB_shader_stencil_export
GL_ARB_shader_storage_buffer_object
GL_ARB_shader_subroutine
GL_ARB_shader_texture_image_samples
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shading_language_420pack
GL_ARB_shading_language_include
GL_ARB_shading_language_packing
GL_ARB_shadow
GL_ARB_spirv_extensions
GL_ARB_stencil_texturing
GL_ARB_sync
GL_ARB_texture_barrier
GL_ARB_texture_border_clamp
GL_ARB_texture_buffer_object
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_buffer_range
GL_ARB_texture_compression
GL_ARB_texture_compression_bptc
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_cube_map_array
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_filter_minmax
GL_ARB_texture_float
GL_ARB_texture_gather
GL_ARB_texture_mirror_clamp_to_edge
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_multisample
GL_ARB_texture_non_power_of_two
GL_ARB_texture_query_levels
GL_ARB_texture_query_lod
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_stencil8
GL_ARB_texture_storage
GL_ARB_texture_storage_multisample
GL_ARB_texture_swizzle
GL_ARB_texture_view
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_transform_feedback_instanced
GL_ARB_transform_feedback_overflow_query
GL_ARB_transpose_matrix
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_attrib_binding
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_vertex_type_10f_11f_11f_rev
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ARB_window_pos
GL_ATI_blend_equation_separate
GL_ATI_draw_buffers
GL_ATI_fragment_shader
GL_ATI_separate_stencil
GL_ATI_texture_compression_3dc
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXT_EGL_image_storage
GL_EXT_EGL_sync
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_compiled_vertex_array
GL_EXT_copy_texture
GL_EXT_direct_state_access
GL_EXT_draw_buffers2
GL_EXT_draw_instanced
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_packed_pixels
GL_EXT_pixel_buffer_object
GL_EXT_point_parameters
GL_EXT_polygon_offset_clamp
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shader_framebuffer_fetch_non_coherent
GL_EXT_shader_integer_mix
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_subtexture
GL_EXT_texture
GL_EXT_texture3D
GL_EXT_texture_array
GL_EXT_texture_buffer_object
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_latc
GL_EXT_texture_compression_rgtc
GL_EXT_texture_compression_s3tc
GL_EXT_texture_cube_map
GL_EXT_texture_edge_clamp
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_env_dot3
GL_EXT_texture_filter_minmax
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_object
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_R8
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shadow_lod
GL_EXT_texture_shared_exponent
GL_EXT_texture_snorm
GL_EXT_texture_swizzle
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array
GL_EXT_vertex_array_bgra
GL_IBM_multimode_draw_arrays
GL_IBM_rasterpos_clip
GL_IBM_texture_mirrored_repeat
GL_INGR_blend_func_separate
GL_KHR_blend_equation_advanced
GL_KHR_context_flush_control
GL_KHR_debug
GL_KHR_no_error
GL_KHR_parallel_shader_compile
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_KHR_texture_compression_astc_ldr
GL_KHR_texture_compression_astc_sliced_3d
GL_MESA_pack_invert
GL_MESA_shader_integer_functions
GL_MESA_texture_signed_rgba
GL_MESA_window_pos
GL_MESA_ycbcr_texture
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_copy_depth_to_color
GL_NV_copy_image
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_half_float
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
</details>

### 3D printer brand / version + firmware version (if known)
PRUSA MK3S+ but the printer is not the problem

### Behavior
I updated to slicer 2.4.1 and the file that I print a lot, which is 33 megs, is now dramatically slower in the slicer.
Even trying to change from "3D editor view" to "preview" takes several seconds.
The "DELETE ALL" button takes several seconds.
The weird thing is even going back to 2.4.0 it is now slower.  This may have something to do with the configuration upgrade but I am not sure.
The problem appears to be with this particular file but it worked fine in previous versions.

_Is this a new feature request?_  NO

#### Project File (.3MF) where problem occurs
the .3MF file is 33 megs and it would not upload.  please contact me at msproul -at- skychariot.com and I will send you the file.

lukasmatena, 2022-03-14:
If you also see the slowdown in 2.4.0 which, as you claim, worked before, it is probably not caused by a change in the software. IMHO it can either be caused by something in the configuration or for example by your system/GPU driver update or something. It is easy to rule out the configuration - just backup the configuration folder (do NOT delete it) and start slicer fresh. If the problem does not disappear, it is quite likely on your end. If it does disappear, check that it reappears after you delete the new configuration folder and replace it with the backup. Thanks.

lukasmatena, 2023-02-06:
No response, closing. The "2.4.0 is now also slower" part indicates that the problem is not in PrusaSlicer.

closed 2023-02-06 by lukasmatena