prusa3d/PrusaSlicer #8059: Label objects issues

RoboMagus, 2022-03-12:
### Version
2.4.0+win64
Build: PrusaSlicer-2.4.0+win64-202112211614

### Operating system type + version
Windows 10 64bit (build 19044)

### 3D printer brand / version + firmware version (if known)
Prusa MK3s, but irrelevant...

### Behavior
I have found a couple of minor issues when dealing with multiple objects (completed individually) and `Label objects` enabled. 
* when adding multiple identical objects to the plate, their object names in GCode cannot be distinguished. I would expect at least a differentiating factor such as a copy number. However this is only the case when adding multiple _instances_ of the same object.
* Secondly, when manually renaming the items in the right sidebar, this will not flag for reslicing before export. Thus in the output file the objects will keep their old, in my case duplicate, names.

These are easily overcome when you know how slicer outputs its GCode, but when unaware this can be annoying to deal with using e.g. the Cancel Objects plugin,  when because of this only a single object can be found.

_Steps to reproduce:_
1. Add an object to the buildplate. 
2.
  - Increase the instance count a number of times. In the sidebar a subtree shows with the instances of the same object. To reorder, drag an instance out of this subtree, making it a regular old copy.
  - Or instead just copy paste the same object a number of times
3. Now the sidebar will show a number of objects with the same name.
4. Export GCode, find that all objects have identical names.
5. Rename each object in the sidebar.
6. Export again, find that the object names in the GCode have not been changed .(The export name wil automatically update though).
7. To actually get the updated names, reorder the build plate a bit to force reslicing. When exporting now, the names are updated in the GCode labels.

lukasmatena, 2022-03-14:
You are probably right, the GCode export should be invalidated when an object name changes.

tverett, 2023-04-21:
yeah this bit me today. I couldn't use the cancel object plugin when I had a catastrophic failure due to the fact that I had made multiple copies of the object on the plate in prusaslicer using the copy paste keyboard shortcuts vs add instances menu item.  the id and copy numeric are appended to the end of the object string regardless of how you end up with multiple copies of an object so PS just needs to actually make sure they're always unique.

lukasmatena, 2024-02-09:
The first part of the problem (missing invalidation of G-code export) is fixed in [2.7.2-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.2-alpha2).

Regarding the naming, instances are labelled using a postfix after the object name. Objects created by copy-paste are copy-pasted including the name. We may discuss internally what may be best thing to do. I guess that this is now a feature request asking to rename object during copy&paste, am I right?

docktordreh, 2024-11-22:
It would be really convenient if the label "generator" appended a uuid (may as well just be a unique number counting from 1 to the number of objects) to the object names.
From a user perspective, I think the label names should be unique. It wasn't obvious to figure out the reason why mainsail decided to exclude three objects when I just wanted to exclude one.

My use case is printing n instances of a model in m different sizes, e.g 5 instances at 100%, 10 instances scaled at 80%, 10 instances at 60%.

zortje, 2025-02-05:
It took me a while to figure out why my exclude objects didn't work in Klipper. The slicer should really generate unique names, especially since when I am printing multiple objects, it is often the same model repeated and then the name is the same.

Having to remember to rename the objects before printing is tiresome and something the slicer should handle.

synthead, 2025-05-06:
I run into this problem often.  Renaming objects by hand is a chore that I often forget.  When a print of a single object begins to fail, there's a good chance that the unstoppable spaghetti blob will destroy adjacent objects.

I propose adding a simple human-readable suffix to the object names.  For example, say we have this single object:

- `Hinge`

If this gets copied, the new objects could be renamed, like so:

- `Hinge`
- `Hinge (2)`
- `Hinge (3)`

We could also do a simple regex search on the existing object names, so that if `Hinge (3)` is copied, it'll assume that it was originally called `Hinge`, and will assign the next unique suffix to the the object name, like `Hinge (4)`.

Since we're talking about object names that interact with user input, it wouldn't necessarily be exact or fit all use cases.  However, I think there's value in picking a pattern and sticking to it.  We could also allow customizable suffixes (i.e. a printf pattern like ` (%i)`), and an option to turn this feature off.

closed 2026-09-29 by prusaslicer-bot[bot]