prusa3d/PrusaSlicer #8060: [bug] Clean while retracting (parameter wipe[0]) has no effect from 2.4.0

eurohell, 2022-03-12:
### Version
from 2.4.0

### Operating system type + version
Windows 10 64bit

### 3D printer brand / version + firmware version (if known)
Original MK3S+ with MMU2S

### Behavior
Move/Clean while retracting ( wipe[0] )don't work anymore from 2.4.0. Checking the checkbox in and out doesn't change anything. In 2.3.0 when you print a single contour the print head moved a bit further than the end of the contour to wipe the filament onto the object. This don't work anymore and leads to little gaps at the beginning of the contour. 

[image]
[image]
[image]

lukasmatena, 2022-03-14:
I cannot reproduce it, or maybe I don't understand. Please, post a 3MF that works differently in 2.3.0 and 2.4.0. Thanks.

12oclocker, 2022-03-15:
this may be related to the bug I just found, see this post here...
https://github.com/prusa3d/PrusaSlicer/issues/8072

eurohell, 2022-04-05:
[Testfile.zip](https://github.com/prusa3d/PrusaSlicer/files/8421187/Testfile.zip)

This file acts slices different, focus on completing the ring and moving to the next instance, the "wiping" acts different (long wipe on 2.3.0 and almost no wipe at 2.4.0, leads to gaps in the printed parts)

lukasmatena, 2024-02-20:
This should be fixed as of 2.7.2-alpha2. Closing.

closed 2024-02-20 by lukasmatena