prusa3d/PrusaSlicer #8061: issue with slicer not including text

sksharp2, 2022-03-13:
2.4.1 Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_windows 10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known) Prusa Mini +
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_3D model of a part with text standing proud on the front surface. STL file shows engraving but slicer doesn't. Earlier versions of slicer software didn't have this problem.
* _Steps needed to reproduce the problem_Slice a part with text.
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
  * part 
[image]
STL file on build stage
[image]
part after slicing
[image]
Note that the engraving is gone.

_Is this a new feature request?_ Yes

#### Project File (.3MF) where problem occurs

_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

eurohell, 2022-03-13:
Seems like the extrusion width is too big for your engraving. Tried with a font with thicker lines or making the font a bit bigger?

sksharp2, 2022-03-13:
Thanks for your input. I will try your suggestions. I found that if I cut most of the part away the engraving prints. Odd but maybe a clue. 

Steve.

dartrax, 2022-03-13:
> Earlier versions of slicer software didn't have this problem.

Did you try with the same model and same settings in an earlier version? If yes, which one and please provide the .3mf file.

lukasmatena, 2022-03-14:
@sksharp2 Post a 3MF, thanks.

rtyr, 2022-03-30:
No response. The text is visibly too small to be printed with default extrusion width. Closing.

closed 2022-03-30 by rtyr