prusa3d/PrusaSlicer #8062: [Feature Request] Place new model at random position within the print are

ChristophvdF, 2022-03-13:
### Version
2.4.1 and future

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3S

### Behavior wish
Random position of a part.

To prevent the print bed from losing its grip in one particular spot, it would be nice if a new added part could be placed at a random position with in the print area. On my smooth bed the center has lost its stickiness due to printing a lot of parts on it. So I started placing the models all over the print bed. If the slicer could assist me in this it would be a lot of help.

_Bonus_
The slicer could remember the last 4 position and try to avoid these if possible.

_Is this a new feature request?_
Yes

dartrax, 2022-03-13:
Duplicate of #7860, #320

ChristophvdF, 2022-03-13:
Sorry, should have used the search function first before opening up the FR. I will close the issue than. Thank you!

closed 2022-03-13 by ChristophvdF