prusa3d/PrusaSlicer #8064: inconsistent/incorrect support (with vertical gaps ) sometimes generated

netweaver1970, 2022-03-13:
### Version
2.4 based.

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Voron 2.4

### Behavior
generated supports have holes/skips in them, vertical zigzags are connected while they shouldn't be and are not connected when they should have been. It looks like small vertical segments are 'shifted' between zigzag walls and a sort of vertical checker pattern is generated as a consequence.
[image]

It doesn't happen on all generated request, if it happens it's near the object itself, never near the ground.
[image]

It was originally seen in in SS based on PS 2.4, it was investigated there and this piece of code (and bug) originates from PrusaSLicer 2.4, hence the request to create the issue here.

For reference (https://github.com/supermerill/SuperSlicer/issues/2525#issuecomment-1066189978)  

Thanks.

lukasmatena, 2022-03-14:
Can you please provide a 3MF and describe exactly where the problem is, so it is clear what we are talking about? Thanks.

netweaver1970, 2022-03-14:
sure, attached. + screenshot of the issue with this file: 
[image]
[spiral.zip](https://github.com/prusa3d/PrusaSlicer/files/8244030/spiral.zip)

n8bot, 2022-03-15:
This seems like normal behaviour. The support is changing from a zig to a zag where you highlight it.

The connections of the lines are not guaranteed to be in any specific orientation, whereas the lines themselves stick to the placement specified.

closed 2022-11-04 by netweaver1970