prusa3d/PrusaSlicer #8065: Consider supporting zero compression in 3MF

TomasHubelbauer, 2022-03-14:
Hi, I'd like to request that PrusaSlicer gets a new setting to disable compression in 3MF. I'm looking to version my projects in Git and 3MF is a binary format which makes the diffs meaningless.

According to 3MFConsortium/spec_core#61, zero compression is valid in OPC so I believe this should be possible.

Currently I've resorted to versioning ASCII STL from OpenSCAD and just taking note of PrusaSlicer settings. I'm also experimenting with extracting the 3MF file to a directory using a Git hook, but those don't work everywhere (Github web UI) and one can forget to configure then when setting up a repo, so it's not ideal.

Thank you

bubnikv, 2022-04-08:
3MF is always a ZIP, albeit some of the files may be stored into ZIP archive without compression. Is this what you are asking for?

TomasHubelbauer, 2022-04-08:
Yep. My understanding is that this is the closest way to get 3MF to be version control friendly without Git hooks or other scripts to unpack the archive. Of course the headers and the trailing entries of the archive will remain binary, but the individual blobs of the files would be plain text so diffing would be clunky but usable. That's my goal.

bubnikv, 2022-04-08:
Frankly I don't think that git tools would handle diffs of such ZIP files
gracefully. I bet you may try to repack 3MFs to uncompressed ZIPs and try
that uncompressed zip workflow first.

TomasHubelbauer, 2022-04-08:
I do that currently, but it is not very convenient and I'm hoping to find a way that won't rely on my memory or Git hooks to always work. I'm considering also just exporting the slicer settings and the geometry in STL separately if I can't make a usable 3MF workflow work, but it seems that 3MF is the way of the future and it is the PrusaSlicer default, so I'm willing to make compromises to make this work. I reached out to the 3MF org to see if a plain text version of the format is on the roadmap, but it is not and I think it will always be only a binary format.

TomasHubelbauer, 2022-04-08:
I tried it and even though Git sometimes has no problem diffing semi-binary files (like PDFs), with ZIP, even if "store" compression mode is used, the ZIP archive's binary header is too large and the Git binary file detection will see that in the leading slice of the file it checks to determine whether a file is a binary or textual, the majority of the contents appears non-textual and will treat it as a binary still.

Details here https://github.com/TomasHubelbauer/3mf-zip-uncompressed

It is possible to coerce Git to treat it as text and diff it, but it's too far off the default path for me, so I'll stick with unpacking my 3MFs to a directory or perhaps switch back to textual STL. Thank you

closed 2022-04-08 by TomasHubelbauer