prusa3d/PrusaSlicer #8068: Bed temp resets to 0 at start of print

Sonicboom247, 2022-03-14:
### Version
2.4.1 win64

### Operating system type + version
Win 10 x 64

### 3D printer brand / version + firmware version (if known)
Artillery Sidewinder with BTT SKR2 board Marlin 2.0.9.3 bugfix

### Behavior
* everything is fine until custom start gcode is handed off to print gcode, then the bed resets temp to 0 instead of 65 from 70
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_ bed should drop temp to 65 from 70 but instead goes to 0
[test.zip](https://github.com/prusa3d/PrusaSlicer/files/8246173/test.zip)

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

neophyl, 2022-03-14:
gcode is of limited use.  The devs are going to need a saved project file (File>Save project as) from PS to debug things, as requested in the instructions when opening a new issue.

Sonicboom247, 2022-03-14:
my apologies, here you go.  my start routine actually consisted of these additional lines but i removed then in my troubleshooting efforts
; PrusaSlicer start gcode for Artillery Sidewinder x1
; Last updated 3132022
; Set coordinate modes
G90 ; use absolute coordinates
M83 ; extruder relative mode

; Set print temps
M117 Heating Nozzle & Bed
M104 S[first_layer_temperature] ;set extruder temp
M140 S[first_layer_bed_temperature] ; set bed temp
M109 S[first_layer_temperature] ; wait for extruder temp

[Bronco_Logo test.zip](https://github.com/prusa3d/PrusaSlicer/files/8246515/Bronco_Logo.test.zip)

Sonicboom247, 2022-03-14:
even odder now, resliced on my PLA PLUS printer with no BL Touch and Mesh bed leveling using Cura and no temp reset but that is my totally stock go to printer when all else fails. print is an 1:45 so soon as it's done i will re-slice with Prusa and see.  
this is the start gcode for the other printer, 
G28; home all axes
G1 Z2.0 F3000 ; Move Z up to prevent scratching
G0 Z1.0 X0 Y0
M420 S1
M117 Purging extruder
G92 E0    ;Reset extruder
G1 Z2.0 F3000 ; Move Z up to prevent scratching
G1 X0.2 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.2 Y200.0 Z0.2 F750.0 E20 ; Draw priming line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z up to prevent scratching

neophyl, 2022-03-14:
With the slic3r based slicers its normal to explicitly add the heating commands.  If you don't then PS will add really basic ones but they are likely to be where you want them, it just sticks them in there at the start.

If you run the PS printer wizard there is a profile for the X1 in there, install it and compare it to yours for reference.

If your printer is setting the bed to zero with the normal set temperature commands included in the start gcode block then something weird is going on as gcode as sliced like that does not contain any commands to set the bed or extruder to zero after those initial ones except for the ones in the end gcode block.  There is only the change temp gcode to set the extruder to 232 as defined in your filament other layers.

Sonicboom247, 2022-03-15:
closing as using the built in start gcode fixed the issue. no idea what it didn't like about mine but it is leaner now. thanks for the help!

closed 2022-03-15 by Sonicboom247