prusa3d/PrusaSlicer #8071: Bridge Gap Feature for Better Bridges.

12oclocker, 2022-03-15:
To make bridging work better, especially for PETG and other sticky filaments, their should be a "Bridge Gap" feature, similar to how infill works, so you can "Space" the print lines, a settings of 50% would be like 50% infill, giving a full line gap between each bridge line. Bridge lines can stick to the other bridge lines during printing of PETG or sticky filaments, it tears them down, by spacing the first run with gaps each bridge line will stay up, and not get torn down. It results in superior bridges with very little if any distortion, even on tricky filaments like PETG. And the method will work for any filament, so should be considered as a default with 50% as a default value.

### Version
_Version of PrusaSlicer used goes here_
2.4.1

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

dartrax, 2022-03-15:
Reminds me of this FR: #4227
It‘s basically the same with the only difference that after printing the „50% bridge“ like you imagined, in a second pass after a cool down time the remaining 50% will be printed.

Sonicboom247, 2022-03-17:
wondering if this is what is causing the issues with supports with PETG, my top layer of the support seems fused to the bottom of the piece. my guess is the PET profile really slows down the fan, not sure if setting fan for supports a bit higher might help but i have been wracking my brain and filament trying to figure it out...

Sonicboom247, 2022-03-19:
i think this is definitely an issue with 4.2.1 not sure if this is a typo or the setting is causing conflicts. 
[image]

closed 2026-09-29 by prusaslicer-bot[bot]