prusa3d/PrusaSlicer #8073: M112 while try to print with 2.41

ccomtest, 2022-03-15:
### Version
2.4.1

Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
MK3S, 3.10

### Behavior
Since Version 2.41 i have M112 Thermal Runaway always at same layer / time.
Tried this 5 times now. With 2.4.0 i don't have this error. Same file.
Don't know this is due to G92 E0 which i had to use since 2.4.0.

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
3MF cant be uploaded - not a supported format.
[Voron_Design_Cube_v7.zip](https://github.com/prusa3d/PrusaSlicer/files/8251955/Voron_Design_Cube_v7.zip)

rtyr, 2022-03-15:
The g-code output from 2.4.1 is identical to 2.4.1rc1 with your 3MF project. G92 E0 command is not related to thermal regulation, also this command was always included in all our profiles.

You can try to print it without Octoprint. If the error still occurs, the printer should tell you more precisely what thermal runaway type was triggered (bed/hotend). 

You can also check firmware issues on this topic. It seems the error is falsely triggered by Octoprint at least in some cases.
https://github.com/prusa3d/Prusa-Firmware/issues?q=is%3Aissue+is%3Aopen+M112.

Looking at your settings changes, I can see you decreased "disable fan for first" to only 1 layer. This change alone can contribute to thermal runaway, because there is bed temperature change 105->110 at the second layer together with enabled print fan.

ccomtest, 2022-03-15:
Hi, thanks for the answere.
As I don't have this issue with 2.4.1 RC1 and only with 2.4.1 final, I am not sure how Octoprint is a reason for this. At the moment I print with RC1 again. No issues since 3 prints now. With final I was not able to get a print higher than layer 5.

I am always printing with fan off first layer and never had an issue with it (ABS/ASA here only).

So I will try to print from SC card and with fan always on for first layer too, to check this is the issue or not.

rtyr, 2022-03-15:
> As I don't have this issue with 2.4.1 RC1 and only with 2.4.1 final,

As I mentioned, there is no difference between 2.4.1rc1 gcode and 2.4.1 gcode with your project. Only different line is the first one, where is the info about PrusaSlicer version and date/time.

> About that G92 was in all profiles, I don't know this. I had to include this command at layer change since 2.4.1 or I got layer shifts. I did not had to use it before and with 2.4.0. I don't remember this was in the profile before. But I use the same machine profile since Version 2.3.0...

It was added roughly 4 years ago to our stock profiles, but you are using some custom profile. It is not related to your issue anyway, it only resets extruder position to 0.

> I am not sure how Octoprint is a reason for this

M112 (emergency stop) command was sent by Octoprint (that is how I knew you are using Octoprint). But it can still be valid thermal runaway situation, caused by damaged thermistor wires, etc. Printing from SD card should point you to the right direction. It will either not fail at all or it should at least give you better error message. But we saw cases in the past, where Octoprint sent M112 command just because some internal Octoprint errors related to lost communication with the printer.

Anyway, I don´t see any issue related to the PrusaSlicer. You may contact our tech support via live-chat at www.prusa3d.com.

ccomtest, 2022-03-16:
Hi,

thanks for your answere.
I understand that the gcode is the same. I don't know why I have this issue only with 2.4.1 and not with 2.4.1 RC1.

I will stick to RC1 and will switch to Klipper and Mainsail. This should fix it then.

Thermistor is new. And if this would be a cable issue, I should not have this with only one version I of a slicer i think.

But I will close this issue, if no other has this.
Thanks for your answers.

closed 2022-03-16 by ccomtest