prusa3d/PrusaSlicer #8074: ARM64 build

Gisi0, 2022-03-15:
Will there be an arm64 build only for major releases or also by minor releases?

iluminae, 2022-03-15:
I too was looking for aarch64 builds (AppImage) and saw they were not built for the 2.4.1 release - but two x86_64 builds were made - was this a mistake?

bubnikv, 2022-03-16:
Partially mistake, partially because we don't have the ARM64 builds tightly integrated in the build system. Partly because I wondered how many people will be complaining if these builds are missing.

I will look into that.

iluminae, 2022-03-16:
I think you will see a growing interest in aarch64 builds as the raspberry pi org has finally blessed 64bit OSes, but who knows. I, for one, was extremely excited to see aarch64 appimages for the last version.

bearclaws8, 2022-04-08:
I too was disappointed to see an aarch64 build dropped for 2.4.1.  Myself and several friends are running Pinebook Pros for some lightweight CAD and 3D printing, having the slicer available on them with the 2.4.0 build was a great benefit.  It made for a great compliment to our Mk3S+ as we could export the code right onto the microSD in the laptops reader and pop it into an adapter to put right into the printer.  I would very much look forward to seeing a new aarch64 build in the future.

iluminae, 2022-04-27:
Was this dropped forever? It was announced in the release notes of 2.4.0 (from beta 4 on) as a new supported architecture. Now two releases with this missing.

bubnikv, 2022-04-28:
I worked on the build yesterday, it takes some time to finish. We should
have it ready today.

bubnikv, 2022-05-23:
> Thanks for the builds. Unfortunately both failed on my Manjaro install. Is
> it possible to get it via an AppImage like like the last version?
> Also,sorry for following up so many days later.

We had an issue building the appimages for ARM 64 bit. The AppImages were
built, but they did not launch on our testing R-PI 64 machine, thus we only
distributed the non-AppImage binary builds. We have little control over the
AppImage builder. The AppImage builder adds a binary loader, which was no
more compatible for some reason with our R-PI 64bit testing machine. We did
not have time to investigate the issue further.

Actually the AppImage contains the non-AppImage binaries, thus if they do
not run, the AppImage would not run either. We tested these binaries on our
R-PI 64bit machine and they worked fine.

bearclaws8, 2022-05-24:
@bubnikv Thanks for following up.  I'll see what I can resolve on my end but wanted to provide the error output I'm getting when attempting to install.  If you need any additional information please feel free to reach out.

>sudo pacman -U ~/Downloads/PrusaSlicer-2.4.2+linux-aarch64-GTK3-202204281113.tar.bz2
loading packages...
error: missing package metadata in /home/pine1274/Downloads/PrusaSlicer-2.4.2+linux-aarch64-GTK3-202204281113.tar.bz2
error: '/home/pine1274/Downloads/PrusaSlicer-2.4.2+linux-aarch64-GTK3-202204281113.tar.bz2': invalid or corrupted package

I did like having the AppImage because it helped me get around some of the quirks of using this aarch Manjaro distribution on this hardware.  Unfortunately there is not a robust Ubuntu solution for me to run else I think this would be less of an issue.  Thanks.

bubnikv, 2022-05-24:
IMHO you cannot install the package using pacman.

You need to unpack the archive with

tar xfvj
/home/pine1274/Downloads/PrusaSlicer-2.4.2+linux-aarch64-GTK3-202204281113.tar.bz2

then start prusaslicer from that directory.

bearclaws8, 2022-05-24:
you were correct, thank you.

ggppjj, 2022-09-02:
Is this the proper issue to inquire about Windows ARM64 builds also? The standard binary does work mostly fine with only one weird crash once on my WoA ThinkPad X13s, but a native build seems doable based on the Linux build work being done.

ErwanAliasr1, 2023-06-28:
Would be cool to get an AARCH64 build for Linux

lukasmatena, 2023-06-28:
We will likely provide the builds in near future, I will send an update into this thread when they are online.

Just of of curiosity, could you guys please tell us what devices you plan to run them on?

Gisi0, 2023-06-28:
I run it on a Raspberry 4, RaspberryOS 64bit

bearclaws8, 2023-06-28:
I'll be running on a pinebook pro with a Manjaro OS.

ErwanAliasr1, 2023-06-29:
I'm using Fedora 38 with AsahiLinux Edge on a Mac M1 Mini 2020.
If you have a build procedure I can try it on my system.

bubnikv, 2023-06-29:
We are running the build on R-PI 64bit now, it will take around a day.

ErwanAliasr1, 2023-06-29:
Thanks ! Much appreciated.

Gisi0, 2023-07-18:
Works well,

thank you

ErwanAliasr1, 2023-07-18:
Works well ? I didn't saw a reply to your comment. Is there any release?

Gisi0, 2023-07-19:
Yes, look in release 2.6

lukasmatena, 2023-07-20:
Sorry for the late notification, it is holiday season. As mentioned, the 2.6.0 aarch64(=arm64) builds are available [here](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0).

ErwanAliasr1, 2023-08-16:
It works nice, thx for the aarch64 build.
Will it be a default arch for the incoming releases ?
To me, this issue can be closed.
Thx for considering this ticket.

ErwanAliasr1, 2023-08-23:
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.1-rc1 is not having any aarch64 build, can it be set as a default target please ?

docgalaxyblock, 2023-09-21:
> We will likely provide the builds in near future, I will send an update into this thread when they are online.
> Just of of curiosity, could you guys please tell us what devices you plan to run them on?

@lukasmatena *A bit late but* on the Windows side I am using a Lenovo IdeaPad 4G (14Q8C05) and would really love if winarm64 builds will be considered in the future.

and on the Linux side a Jetson Nano upgraded to 20.04 that successfully runs PS 2.6.0 GTK2&3 Appimages.
sadly on 2.6.1 they are not available yet. Could they please be added?

_If it is caused by some build error could you please note in future releases why they are missing?_

closed 2023-12-18 by Gisi0

ggppjj, 2023-12-18:
Is there a Windows Arm64 build in the works?

Gisi0, 2023-12-18:
> Is there a Windows Arm64 build in the works?

I don't know, from my side this issue was the linux build for arm64.