prusa3d/PrusaSlicer #8075: Build warnings

ArchangeGabriel, 2022-03-15:
### Version

2.4.1

### Operating system type + version

Arch Linux

### Behavior

I’ve read in the changelog that you try to compile warnings free, so here are a bunch of them with an up to date toolchain:
```
In file included from /build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/clipper/clipper_z.cpp:7:
/build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/clipper/clipper.cpp: In function ‘void ClipperLib_Z::InitEdge(ClipperLib_Z::TEdge*, ClipperLib_Z::TEdge*, ClipperLib_Z::TEdge*, const IntPoint&)’:
/build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/clipper/clipper.cpp:448:14: warning: ‘void* memset(void*, int, size_t)’ clearing an object of type ‘struct ClipperLib_Z::TEdge’ with no trivial copy-assignment; use assignment or value-initialization instead [-Wclass-memaccess]
  448 |   std::memset(e, 0, sizeof(TEdge));
      |   ~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
In file included from /build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/clipper/clipper.cpp:41,
                 from /build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/clipper/clipper_z.cpp:7:
/build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/clipper/clipper.hpp:220:10: note: ‘struct ClipperLib_Z::TEdge’ declared here
  220 |   struct TEdge {
      |          ^~~~~
/build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/admesh/normals.cpp: In function ‘void stl_fix_normal_directions(stl_file*)’:
/build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/admesh/normals.cpp:196:29: warning: unused variable ‘temp’ [-Wunused-variable]
  196 |                 stl_normal *temp = head->next;  // Delete this facet from the list.
      |                             ^~~~
```
```
/build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/admesh/shared.cpp: In function ‘void stl_generate_shared_vertices(stl_file*, indexed_triangle_set&)’:
/build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/admesh/shared.cpp:106:55: warning: comparison of integer expressions of different signedness: ‘int’ and ‘uint32_t’ {aka ‘unsigned int’} [-Wsign-compare]
  106 |                                 } else if (next_facet == facet_idx) {
      |                                            ~~~~~~~~~~~^~~~~~~~~~~~
/build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/admesh/shared.cpp: In function ‘bool its_write_off(const indexed_triangle_set&, const char*)’:
/build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/admesh/shared.cpp:142:27: warning: comparison of integer expressions of different signedness: ‘int’ and ‘std::vector<Eigen::Matrix<float, 3, 1, 2> >::size_type’ {aka ‘long unsigned int’} [-Wsign-compare]
  142 |         for (int i = 0; i < its.vertices.size(); ++ i)
      |                         ~~^~~~~~~~~~~~~~~~~~~~~
/build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/admesh/shared.cpp: In function ‘bool its_write_vrml(const indexed_triangle_set&, const char*)’:
/build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/admesh/shared.cpp:176:22: warning: comparison of integer expressions of different signedness: ‘int’ and ‘std::vector<Eigen::Matrix<float, 3, 1, 2> >::size_type’ {aka ‘long unsigned int’} [-Wsign-compare]
  176 |         for (; i + 1 < its.vertices.size(); ++ i)
      |                ~~~~~~^~~~~~~~~~~~~~~~~~~~~
```
```
/build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/admesh/stlinit.cpp: In function ‘bool stl_read(stl_file*, FILE*, int, bool)’:
/build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/admesh/stlinit.cpp:206:39: warning: ‘void* memset(void*, int, size_t)’ clearing an object of type ‘stl_normal’ {aka ‘class Eigen::Matrix<float, 3, 1, 2>’} with no trivial copy-assignment; use assignment or value-initialization instead [-Wclass-memaccess]
  206 |                                 memset(&facet.normal, 0, sizeof(facet.normal));
      |                                 ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/eigen3/Eigen/Core:295,
                 from /usr/include/eigen3/Eigen/Geometry:11,
                 from /build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/admesh/stl.h:31,
                 from /build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/admesh/stlinit.cpp:33:
/usr/include/eigen3/Eigen/src/Core/Matrix.h:178:7: note: ‘stl_normal’ {aka ‘class Eigen::Matrix<float, 3, 1, 2>’} declared here
  178 | class Matrix
      |       ^~~~~~
/build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/admesh/util.cpp: In function ‘void stl_repair(stl_file*, bool, bool, bool, float, bool, float, bool, int, bool, bool, bool, bool, bool, bool)’:
/build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/admesh/util.cpp:327:9: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
  327 |         if (! tolerance_flag)
      |         ^~
/build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/admesh/util.cpp:329:17: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
  329 |                 if (! increment_flag)
      |                 ^~
/build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/admesh/connect.cpp: In function ‘void stl_check_facets_nearby(stl_file*, float)’:
/build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/admesh/connect.cpp:487:48: warning: comparison of integer expressions of different signedness: ‘int’ and ‘uint32_t’ {aka ‘unsigned int’} [-Wsign-compare]
  487 |         if (stl->stats.connected_facets_3_edge == stl->stats.number_of_facets)
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/admesh/connect.cpp: In lambda function:
/build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/admesh/connect.cpp:536:60: warning: comparison of integer expressions of different signedness: ‘int’ and ‘uint32_t’ {aka ‘unsigned int’} [-Wsign-compare]
  536 |                                         if (other_face_idx != stl->stats.number_of_facets) {
      |                                             ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/admesh/connect.cpp: In instantiation of ‘void HashTableEdges::insert_edge(stl_file*, const HashEdge&, MatchNeighbors) [with MatchNeighbors = HashTableEdges::insert_edge_exact(stl_file*, const HashEdge&)::<lambda(const HashEdge&, const HashEdge&)>]’:
/build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/admesh/connect.cpp:143:20:   required from here
[... 172 lines cut ...]
                 from /usr/include/boost/asio/detail/deadline_timer_service.hpp:31,
                 from /usr/include/boost/asio/basic_waitable_timer.hpp:22,
                 from /usr/include/boost/asio/steady_timer.hpp:22,
                 from /usr/include/boost/asio/basic_socket_streambuf.hpp:36,
                 from /usr/include/boost/asio/basic_socket_iostream.hpp:24,
                 from /usr/include/boost/asio/ip/tcp.hpp:20,
                 from /build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/slic3r/Utils/TCPConsole.cpp:3:
/usr/include/boost/bind.hpp:36:1: note: ‘#pragma message: The practice of declaring the Bind placeholders (_1, _2, ...) in the global namespace is deprecated. Please use <boost/bind/bind.hpp> + using namespace boost::placeholders, or define BOOST_BIND_GLOBAL_PLACEHOLDERS to retain the current behavior.’
   36 | BOOST_PRAGMA_MESSAGE(
      | ^~~~~~~~~~~~~~~~~~~~
/build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/slic3r/GUI/Search.cpp: In constructor ‘Slic3r::Search::SearchListModel::SearchListModel(wxWindow*)’:
/build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/slic3r/GUI/Search.cpp:720:29: warning: loop variable ‘icon’ of type ‘const string&’ {aka ‘const std::__cxx11::basic_string<char>&’} binds to a temporary constructed from type ‘const char* const’ [-Wrange-loop-construct]
  720 |     for (const std::string& icon : { "cog", "printer", "sla_printer", "spool", "resin" })
      |                             ^~~~
/build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/slic3r/GUI/Search.cpp:720:29: note: use non-reference type ‘const string’ {aka ‘const std::__cxx11::basic_string<char>’} to make the copy explicit or ‘const char* const&’ to prevent copying
```
```
/build/prusa-slicer/src/PrusaSlicer-version_2.4.1/src/avrdude/config.c:354: warning: type of ‘yy_scan_bytes’ does not match original declaration [-Wlto-type-mismatch]
  354 | extern YY_BUFFER_STATE yy_scan_bytes(const char *base, size_t size);
      | 
avrdude/lexer.c:2825: note: type mismatch in parameter 2
avrdude/lexer.c:2825: note: type ‘int’ should match type ‘size_t’
avrdude/lexer.c:2825: note: ‘yy_scan_bytes’ was previously declared here
avrdude/lexer.c:2825: note: code may be misoptimized unless ‘-fno-strict-aliasing’ is used
```

bubnikv, 2022-03-16:
What do you recommend?

bubnikv, 2022-03-16:
Many of the warnings are produced by the 3rd party libraries.

ArchangeGabriel, 2022-03-19:
Well those should be upstreamed I guess. I’m mostly concerned about `-Wmisleading-indentation` ones that look actually serious issues.

kostnik, 2025-03-09:
getting an error on gentoo:

```sh
: && /usr/bin/x86_64-pc-linux-gnu-g++ -O2 -pipe -march=tigerlake -flto=9 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing -fext-numeric-literals -Wall -Wno-reorder -Wl,-O1 -Wl,--as-needed -Wl,-z,pack-relative-relocs    -rdynamic  -Wl,--dependency-file=bundled_deps/avrdude/CMakeFiles/avrdude-slic3r.dir/link.d bundled_deps/avrdude/CMakeFiles/avrdude-slic3r.dir/avrdude/main-standalone.cpp.o -o bundled_deps/avrdude/avrdude-slic3r  bundled_deps/avrdude/libavrdude.a  bundled_deps/liblocalesutils.a && :
/var/tmp/portage/media-gfx/prusaslicer-2.9.0/work/PrusaSlicer-version_2.9.0/bundled_deps/avrdude/avrdude/config.c:354:24: error: type of 'yy_scan_bytes' does not match original declaration [-Werror=lto-type-mismatch]
  354 | extern YY_BUFFER_STATE yy_scan_bytes(const char *base, size_t size);

```

I can build avrdude 7.2 separately without issues

closed 2026-09-29 by prusaslicer-bot[bot]