prusa3d/PrusaSlicer #8076: PrusaSlicer-2.4.1 does not close with no 3D-Model loaded and unsafed

Moerphium, 2022-03-16:
### Version
_Version of PrusaSlicer used goes here_
Version 2.4.1+win64
_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10 Home, 10.0.19044 Build 19044
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_ 
Prusa Mini+

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
Every time i open PrusaSlicer and want to close it right away, i am prompted by the "do you want to safe"-dialog. Since i did not load any model into PrusaSlicer i answer the dialog with "NO". This closes the dialog but not PrusaSlicer. This can be repeated for ever but PrusaSlicer does stay open. I have to close it from the TaskManager in Windows.

This also happens, if i delete any loaded 3D-model from the project and want do close PrusaSlicer without safeing the project. Same dialog prompt "Do you want to safe the Project .3mf?", same answer "NO" --> Dialogprompt closes but Prusaslicer stays open.

Both happens no matter if i klick the menu "File"->"Quit" or use the X in the upper right corner. I even tested it in different language settings (English and German), but i get the same bug.

I would like to close PrusaSlicer  without safeing a project file, as it does if i safe the file. But if my answer is "No" it should also close the program.

_Is this a new feature request?_
no

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

there is no file that is where the problem beginns. I just want to quit PrusaSlicer without safeing.

Sonicboom247, 2022-03-19:
same just happened to me but i made some changes to only the settings without loading a model, was prompted and said no and it did not close, pressed X again was prompted again, and this time saved, and it closed. so it is not recognizing an empty plate for me.

buzzysale, 2022-04-04:
Can confirm this bug exists on Mac also.  Very problematic for files that are changed and the user needs to reopen the file, the only way out is to "save as" a different file name and then it will close.  I believe the behaviour for that "No" button is incorrectly set to "Cancel dialog" instead of "close without saving".

lukasmatena, 2022-04-13:
Should be fixed in [PrusaSlicer 2.4.2-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.2-rc2). Let us know if it isn't. Closing the issue.