prusa3d/PrusaSlicer #8077: PrusaSlicer has encountered a fatal error: "CGAL mesh boolean operation crashed."

gstrrgithub, 2022-03-16:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

gstrrgithub, 2022-03-16:
I failed when try to create hole in my 3D design... how to fix this? what correct position to make hole?

Thank you

rtyr, 2022-03-16:
Please provide your 3mf project and more information about PS version, OS.

bubnikv, 2022-03-16:
Holes could be drilled over model region which is valid: free of self
intersections and without holes.

gstrrgithub, 2022-03-21:
> Please provide your 3mf project and more information about PS version, OS.
I'm using Prusa SL1S speed, for app I'm using PrusaSlicer 2.4.1
[side.zip](https://github.com/prusa3d/PrusaSlicer/files/8312898/side.zip)

cosmith71, 2022-12-13:
I'm having the same issue.

"CGAL mesh boolean operation crashed" error when drilling holes after fixing with Netfabb. It does not occur without Netfabb repair.  Error occurs during slicing.  There is an incidental "WARNING:  Mesh to be hollowed is not suitable for hollowing (does not bound a volume".

PrusaSlicer version 2.5.0, Windows 11

[image]
[wanda_scarlet-witch_legs.zip](https://github.com/prusa3d/PrusaSlicer/files/10219379/wanda_scarlet-witch_legs.zip)

gurux13, 2022-12-28:
Happened to me today too.

[Project zip](https://github.com/prusa3d/PrusaSlicer/files/10314287/bug.zip)
[image]

It does not happen with no hole added (and the warning about the object not bounding a volume is not shown). I don't think the hole is cutting through any geometry though.

2.5.0+win64, Windows 10 22H2 (19045.2364), Prusa SL1S

closed 2026-09-29 by prusaslicer-bot[bot]