prusa3d/PrusaSlicer #8079: version 2.4.1 duel printer all fluffed up

gixxerboss, 2022-03-16:
Constant please add g92 e0 at layer gcode, and wipe tower only supported with relative e distances having to revert back to 2.3.3
### Version
_Version of PrusaSlicer used goes here_ 2.4.1

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand/version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2022-03-17:
From 2.4.1-beta3 release notes:

> In [2.4.1-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.1-beta1), we added a check that G92 E0 is present in layer change G-Code when relative extruder addressing is used. This is needed to prevent error accumulation in firmware. From this version, the check is only applied for Marlin firmware, because as far as we know, other firmware flavors do not suffer from this issue. https://github.com/prusa3d/PrusaSlicer/issues/7846
> 
> Related to the previous improvement: our G-Code preview works by mimicking the Marlin firmware, therefore if the G92 E0 was missing and the error accumulation became visible, it would also be visible in the G-Code preview. Because only Marlin firmware suffers from this issue in reality and there is an explicit check for the G92 E0 when Marlin is selected, we have now modified the G-Code Preview so the loss of precision problem is not visualized. https://github.com/prusa3d/PrusaSlicer/issues/4881, https://github.com/prusa3d/PrusaSlicer/issues/5073, https://github.com/prusa3d/PrusaSlicer/issues/6336

So I guess only thing you need to do is add that G92 E0 command to the before layer change gcode. Wipe tower is currently supported only with relative extrusion mode, no change there (it is same in 2.3.3).

We can check your settings if you provide your 3MF project file.

rtyr, 2022-03-30:
No response. Closing.

closed 2022-03-30 by rtyr