prusa3d/PrusaSlicer #8080: [Feature request] Automatic variable layer height -- specify minimum and/or maximum height

WereCatf, 2022-03-17:
Currently, you don't really get much control over the automatic variable layer height. You can adjust the quality/speed - setting, sure, but what if you don't want it e.g. decreasing the layer height over the currently-selected print setting, since you feel it's good enough as-is? Or what if you don't want the tool going beyond some maximum layer height?

Yes, you can manually attempt to adjust those layers afterwards with the mouse, but that's horribly imprecise and rather annoying. If the tool simply let you specify those limits, you could skip the manual step entirely.

As an example, I typically print models and things with small details at 0.15mm layer height and I don't feel like I benefit from going any lower than that. Alas, the tool doesn't offer me a way of limiting the automatic generation to that value and so I have to manually adjust the result, but that adds a couple of minutes of spent time that could've been used more productively elsewhere.

neophyl, 2022-03-17:
Yes it does in a way. The limits are taken from the settings for your extruder in your printer profile for minimum and maximum layer height.

WereCatf, 2022-03-17:
> Yes it does in a way. The limits are taken from the settings for your extruder in your printer profile for minimum and maximum layer height.

Can't really expect most people to know to look there first, when using the variable layer height tool. If adding the limit-settings in the tool itself is out of the question, then it'd be a good idea to at least add a tooltip or something to point people to the profile's extruder-settings, though, personally, I think that's a clunky approach.

neophyl, 2022-03-17:
Hey I didn't say it was ideal or obvious.  I was just pointing out that the limits ARE actually defined someplace.  It's why my personal printer profile has values of 0.1-0.3 set, even though its capable of lower.

If the function were to be improved I'd like to see an optional step value defined as well as a minimum and maximum.  That way for those whose printers work best with specific increments they would be covered.  

Although given that if you turn on non soluble supports it is also using a free ranging variable layer height for those anyway it would mean a bunch more work and testing as you would be effecting supports also at that point.

Garr-R, 2022-04-22:
Unless there is a specific change request, I'd recommend closing this issue since the feature is already in the program.

WereCatf, 2022-04-23:
Even a tooltip telling the user to look in the printer-settings for minimum and maximum layer height would be an improvement, since I would hazard a guess that many people won't realize that -- I, for example, had not -- but then again, I guess I'm also the only one who cares, so fine.

closed 2022-04-23 by WereCatf

derpue, 2022-05-09:
Although closed, please reconsider adding that to the Variable Layer Height Menu, due to the following reasons.

The Min/Max Values (as well as most other values) in the Printers Profile are - at least to me - considered to be the min/max the (physical) printer/extruder is capaple of and/or the limits i want at to have at most.
For slicing a specific object i often need different limits, depending on the object. So while working on that object, i just want to set it there, if i need to adjust it. 
The values in the printers profile should be uses as defaults and limits when invoking the VLH menu. 
Of course i could switch to the printers profile, change a value there or accidentially mess up the printers profile, switch back to the plater, recalc, maybe go an additional round to the printers settings, and so on. Having it in the menu, where the other parameters for the adaptive algorithm are enters seems just more logical to me.